Ultimaker/Cura #7242: 3mf files don't function the same way as stl and cause issues with laying flat on bed and boundaries and sizing

red913, 2020-03-08:
**Application version**
4.5
**Platform**
Windows 10, Acer Aspire a717-72G-76V1
Core i7, no throttling
Nvidida GeForce GTX 1060
**Printer**
Ender 3
**Reproduction steps**
Took a file that contained many stl parts, loaded into windows 3D builder. Deleted all but 3 parts. Saved that as a new 3mf file. 
Loaded that file into Cura

**Screenshot(s)**
File loaded, untouched,100% size and scale large models off
[image]
selected a single part, size changes to 3.94 but is still at 100%
[image]
Rotated 90 degrees, does not go flat to the bed, just hovers on this side. The opposite side will go to the bed. "...align face to build plate" doesn't work either. 
[image]
The wire framing is treated as the actual size of the model and interferes with placement and recognition as being on the buildplate.
[image]

**Project file**
https://we.tl/t-5jPxJovwDR
stl's are the original files. 3mf files are the one's I created in 3D builder.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4303415/cura.log)

Ghostkeeper, 2020-03-11:
Nice models! You're a talented designer.

The positioning and the bounding box are requirements that we need to keep to for the 3MF specification. 3MF specifies the exact position of the model on the build plate. And stupidly that is also specified from the [front left corner of the build volume](https://3mf.io/3d-manufacturing-format/#_Toc444090304) rather than the middle, so delta printers be damned.

You can work around this by right-clicking on your model and selecting "Ungroup models" (or pressing Ctrl+Shift+G). This will remove the model(s) in your 3MF from the group, does away with the bounding box of the 3MF file and causes it to drop down to the build plate if that preference is enabled.

The "lay flat on face" feature should work on groups too, I think. If you're not used to working with model groups this would otherwise be very confusing. Currently it doesn't work on that, so you can only lay flat one object at a time. For instance you can do the ungrouping as before, or you can Ctrl+click on your model within the group to select just the model itself and then do the lay flat on face. But you shouldn't need to. So I'll classify that as a bug.

nallath, 2020-03-11:
We already have a ticket for the lay flat on our internal backlog (CURA-6794)

red913, 2020-03-15:
Ungrouping solved all my problems.

closed 2020-03-15 by red913