Ultimaker/Cura #7244: The disappearance of the bridge

deniman, 2020-03-08:
**Application version**
4.4.1 and 4.5.0

**Platform**
Windows 7 x64

**Printer**
Creality3D Enrder-3 

**Reproduction steps**
I rotate my model for 90 degrees and  I click on the slice button and get the bridge disappearing in the preview.

**Screenshot(s)**
[image]
[image]

**Actual results**
There is no bridge.

**Expected results**
The model must be slice with all the details.

**Project file**
[CE3_Body1.zip](https://github.com/Ultimaker/Cura/files/4303482/CE3_Body1.zip)

**Log file**
[log.zip](https://github.com/Ultimaker/Cura/files/4303487/log.zip)

fieldOfView, 2020-03-08:
Check your model in X-Ray view. There are two faces that (one on each side of the bridge) that should not be there.

[image]

deniman, 2020-03-08:
In my version of Crua there is no this option.

fieldOfView, 2020-03-09:
Yes it does, though it has been hidden away on the Preview tab:
[image]

closed 2020-03-11 by nallath

Ghostkeeper, 2020-03-11:
You can also try the Extensive Stitching option in the Mesh Fixes category. It's made to handle this case.