Ultimaker/Cura #7250: [4.5] Standard profiles disapears after creating a custome profile

Goodix33, 2020-03-09:
**Application version**
4.5

**Platform**
Windows 10

**Printer**
Creality 3D CR-10

**Reproduction steps**
1. I click on custom to customize printing settings
2. I change some settings
3. I save my changes as a new profile

**Screenshot(s)**
[image]
[image]

**Actual results**
Pre-installed profiles dispears in the profiles list. Only the custom profile is showing up

**Expected results**
I should have pre-installed profiles AND the custom profiles available

**Project file**
N/A

**Log file**
N/A

**Additional information**
Maybe it's a problem with me using Cura but for what I remind in previous version it was working like this

mabus5, 2020-03-10:
This happened to me also. After restarting Cura several times and repeatedly switching profiles, the standard profiles did eventually come back. Unfortunately I didn't document what I did as I went.

Goodix33, 2020-03-10:
I have made some trials on Cura 4.4.1. I have installed this older version, changed 2 or 3 parameters, and it keeps both parameters (default and custom) as expected.
I uninstalled it, remove all the profiles folders in the computer, installed again and tried to change all the parameters I have to change. And after saving the new profile, the default parameters disapeared.
So it's not only an issue with 4.5 version, it's also on 4.4.1 and it seems to append when a large quantity of parameters is changed.

nallath, 2020-03-11:
Could you share your configuration folder and the logs?

PuterPro, 2020-03-11:
I, too have this problem. I loaded a set of profiles from Youtuber Chep - https://www.chepclub.com/cura-profiles.html
After I did that 4.5 had no standard profiles. I went back to using 4.3 and all was well. 

Tried restarting and going into the profiles in 4.5 several times, no joy, in fact, the ADDED profiles were now gone from the drop down, although the ADDED profiles do show in the Profile Manager (odd). Standard profiles are MIA.
Here's my log & config:
[cura_4.5.0.zip](https://github.com/Ultimaker/Cura/files/4321114/cura_4.5.0.zip)

Goodix33, 2020-03-12:
Hi, please find my files attached
[4.5.zip](https://github.com/Ultimaker/Cura/files/4326337/4.5.zip)

Ghostkeeper, 2020-03-13:
Thanks for those log files. Sadly, they didn't provide any clarity on why it doesn't load your profiles. It clarifies though that there is nothing wrong with the syntax of the profiles themselves. Something else is wrong, like the profiles being made for the wrong printer or something like that.

For the rest those zip files don't contain your configuration folders (the folders shown when going Help -> Show configuration folder... in Cura's application menu). They just contain your preferences file. We're looking for the folders called `quality_changes`, `machine_instances`, `quality`, etc.

When I try loading those profiles from Chep's website they work fine:
[image]
The default profiles don't disappear and everything is still listed in the drop-down.

PuterPro, 2020-03-13:
@Ghostkeeper - Well, glad to hear _yours_ is working, shame mine isn't ... LOL
I get CHEP's profiles but the standards are gone. I don't have time right now, but I'll try and do a total nuke of Cura and reload from scratch this weekend.  I'll report back here.
Any thoughts you have on this will be gratefully accepted :-) Thanks for your work so far.

Goodix33, 2020-03-13:
As @PuterPro , i'm glad that it's working for you, but unfortunatelly it's not on my computer.
I have tried 3 times to do a new install from scratch and each time I had the default profiles dispearing after registering a new custom profile with lot of changes.
I send my complete configuration folder. Maybe it will help to understand what is happening.
[complete cura 4.5.zip](https://github.com/Ultimaker/Cura/files/4331520/complete.cura.4.5.zip)

PuterPro, 2020-03-13:
@Goodix33 Not MINE, @Ghostkeeper said HIS was working, mine is still down, I'm using 4.3 as a workaround.
Well now he has two config's maybe he'll see something. I'll try the nuke option tomorrow if I can. Good luck!

PuterPro, 2020-03-14:
Update: I was also experiencing #7267 as I mentioned. Today, chasing _this_ bug, I did a complete wipe of ALL Cura installs (I had 4.3 & 4.5 loaded), after I Exported all Profiles and Materials.

I'm a PC tech so I did a thorough scrub of everything, ALL files, folders (hidden etc) and removed ALL Registry entries for Cura & Ultimaker. FULL uninstall.

I then loaded the latest 4.5 EXE (not msi). Went flawlessly. I then Imported all my Profiles and Materials. Still fine.
Closed, and reopened Cura, and loaded an STL. Still fine. It was set to the Default PLA (Generic) @ 205°C.
When I chose a new Material (Polymaker PC-Max) , and **it did NOT update the 4 Hotend temps**, as reported in the #7267 bug report.

I then changed a PROFILE to a custom one (Fine .12 PLA+ Raft,coasting & Adaptive) and two things happened;
1. The temps changed to the proper Material Temps. (260°C)
2. **The STANDARD Profiles VANISHED, as reported here.**

BUT - When I changed to one of the CHEP profiles [CHEPV4.5_012(Best) ], THE TEMPS WENT BACK TO PLA (205°C). Toggling to MY profile (Fine .12 PLA+ Raft ...) will switch the temps to correct 260°C.

NOTHING I do, however, will bring back the Standard Cura Profiles, they're GONE.

So, the bugs are real, and apparently related.

Here are the Profiles and Material, and STL files I loaded that triggered the behavior.
I filed this same info on the #7267 bug report.
[[4.5.0] Temp & Profile Bug.zip](https://github.com/Ultimaker/Cura/files/4333371/4.5.0.Temp.Profile.Bug.zip)

Ghostkeeper, 2020-03-17:
I tried to follow your steps. This is what I did.

1. Wipe Cura configuration completely. (Cura doesn't place any registry entries. The Windows installers do though to register Cura as being installed so that it appears in the software list. This doesn't influence the runtime of Cura. I'm on Linux anyway so that isn't an issue.)
2. Added CR-10 printer.
3. Imported Chep's "Good" profile. The profile now appears in the list. I didn't take the effort to load all of them this time, sorry.
4. I don't have this material you mention, Polymaker PC-Max, in my Cura installation. Did you create it yourself? Anyway, I selected Generic PC. There is no quality profile for this, so the material selection appears yellow and indeed there are no standard profiles then. There is only Chep's profile. The printing temperature is 270 degrees which is appropriate for PC.
5. Changed the printing speed and saved this as a custom profile. The temperatures are still the same at 270 degrees, which is correct.
6. Switched to Chep's profile. The temperature is now 205 degrees since this profile overrides the temperature.
7. Switched the material back to Generic PLA. The PLA materials now reappear.

From what I can see this is working correctly, but you appear to have a misunderstanding. The profiles you saw are not the "standard profiles". They are profiles specific to the configuration of Creality printers + 0.4mm nozzle + PLA. The Creality profile set contains profiles for [lots of different configurations](https://github.com/Ultimaker/Cura/tree/master/resources/quality/creality/base), but not for PC. There are no profiles for that material, so the list of default profiles should appear empty.

Am I correct in this assessment?

Goodix33, 2020-03-17:
YES !!!
You are absolutely right. Mystery solved.
I change the material back to a generic one and the standard profiles are displayed.

Thanks for your help

closed 2020-03-17 by Goodix33

PuterPro, 2020-03-18:
Wow! Thx @Ghostkeeper for such a great answer, you ROCK!
Yes, PC-Max is a custom material I made for Polymaker PC (they used to call it PC-Max).

Didn't realize the relationship between MACHINE profiles and MATERIAL profiles.
For those of your who were following this, there is another explanation from @nallath in report #7281 where I posted separately on this. There is some more info in #7267 about temps resetting with a custom (downloaded) profile which is tied to this same thing.
It's very cool to get such great support, I know the work involved. Thx again.