Ultimaker/Cura #7252: Unable to slice/Outside build volume

Tom950, 2020-03-10:
**Application version**
4.5.0

**Platform**
Windows10
**Printer**
Ender 3 Pro

**Reproduction steps**
1. Load Stl
2. Try to slice

**Screenshot(s)**
[image]

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
Should slice like previous versions which slice fine with same files

**Project file**
https://we.tl/t-F901Jlg09f

**Log file**
2020-03-09 20:24:17,336 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [F:\Ultimaker Cura 4.5\resources\shaders\transparent_object.shader]...
2020-03-09 20:24:18,311 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles
2020-03-09 20:24:21,327 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [225]: Starting to slice...
2020-03-09 20:24:21,327 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [243]: Going to slice build plate [0]!
2020-03-09 20:24:21,334 - WARNING - [Thread-2] UM.Decorators.deprecated_function [20]: <function GlobalStack.extruders at 0x000001C07F8D7620> is deprecated (since 4.4): Please use extruderList instead.

**Additional information**
Settings in 4.5 are identical to 4.4 and under. Only 4.5 refuses to slice

mahtDFR, 2020-03-10:
Thanks for the report. For some reason the disallowed area for this profile has changed between versions. Generally we don't touch third-party profiles, and I can't see a history of a change made recently by anyone here, so I don't understand how this has happened. Maybe someone else can enlighten us.

The problem is in the definition file. This is located at `<Cura 4.5 installation>/resources/definitions/creality_ender3.def.json`

Find `"machine_disallowed_areas"`

Removing these lines fixes the issue:
`              [[-117.5, 117.5], [-117.5, 108], [117.5, 108], [117.5, 117.5]],
              [[-117.5, -108], [-117.5, -117.5], [117.5, -117.5], [117.5, -108]]`

If you're feeling impatient or these instructions are intimidating, I have done it for you. Just unzip [this file](https://github.com/Ultimaker/Cura/files/4312423/creality_ender3.def.json.zip) in your `<Cura 4.5 Installation>/resources/definitions/` folder, and it should work properly again (at least it did with your model in my tests). This is provided without warranty, etc. Use at your own risk.

See also the related discussion on our forum [here](https://community.ultimaker.com/topic/31403-ultimaker-cura-45-now-available/page/2/?tab=comments#comment-258628).

Tom950, 2020-03-10:
Made the change and back to normal. Thanks!

mahtDFR, 2020-03-12:
There's always a chance that this could break something elsewhere. It's not common practice for us to make changes to contributed profiles, so we won't be putting any dev time into fixing this in master. Outside contributions are welcome if someone wants to look into a robust fix for a future release. I will defer this for now.

augiem94, 2020-03-16:
Actually ran into this myself, reverse engineered the fix. The bounds are there because the Ender 3 may or may not use clips to hold the build surface in place. Maybe a tickbox during machine setup/on the machine page to tell Cura if you are using clips or not?

nallath, 2020-03-16:
> Maybe a tickbox during machine setup/on the machine page to tell Cura if you are using clips or not?

That would be one of the solutions, yeah. But since it's a third-party printer, we won't put in the time to make/fix this.

tvercruysse, 2020-03-30:
@nallath regardless of being a third party printer, just assuming the user did attach *something* to the printer and forcing bounding boxes seems reversed to me. If the user attaches something to the build plate, it should be the user's job to set bounding boxes, not the software. 

Side note: resizing the Ender-3's build volume (X and Y) has no effect to slicing volume. Graphically the build plate extends, but when you place something larger than 220x220, it wil fail to slice. (see image)

[image]

Ghostkeeper, 2020-04-01:
Please see the discussion that led to this change: https://github.com/Ultimaker/Cura/issues/6267 There are multiple people there that disagree with the issue posted here. @kazooless @oofiksoo @jacekk015 @Liger0 do you have opinions about this?

We didn't change this ourselves, but applied the change that was given to us since the result of a previous discussion led us to believe the community found this to be better.

mathiasvr, 2020-04-04:
Please remove these boundaries. It took me hours to figure this out and would be so much easier for a user with "clips" to simply compensate for it during slicing. I didn't read the full discussion but I don't understand how this is reasonable, you cannot even increase your build volume because of these boundaries. At the very least please add a profile for Ender 3 Pro that does not use "clips" by default (it has a magnetic bed).

nallath, 2020-04-05:
The change had nothing to do with clips.

mathiasvr, 2020-04-05:
@nallath Then why?
My Ender works fine without the boundaries so I still don’t understand why they are there?

mahtDFR, 2020-04-06:
the [original commit which caused this](https://github.com/Ultimaker/Cura/commit/726561bf309616107ad874d4e508b3a3ec9e5701) is here. It doesn't explicitly mention clips but there was indeed some discussion about it elsewhere. As this was a community contribution, we'll be looking to the community to make a pull request to fix this if it is decided to be needed - it's not something we really have development time to solve and we are unable to test it because we don't have these particular printers (even less so now that we're all working from home!)

mathiasvr, 2020-04-06:
@mahtDFR I see, thanks. The updated build plate dimensions were indeed correct. Maybe they had previously been set smaller to account for clips, not sure. However, I think it's fairly certain that the added boundaries are there to avoid hitting clips since I cannot find other reasons for them.
I understand and don't expect you to invest considerable time in third party printers.
I am willing to create a pull request to fix this, what is the procedure to ensure community support for the change?
Would it be okay to change the current Ender 3 profile, or may I create a new "Ender 3 Pro" profile.
And BTW this is needed. Currently, the common way to use the full build plate is to use a different slicer. I would like to be able to do this in Cura as well.

Ghostkeeper, 2020-04-07:
The original pull request does mention clips in the commit message: https://github.com/kazooless/Cura/commit/c0acdc3aa724ab7d6f82d639ae76da3e596aab2f

kazooless, 2020-04-07:
Hello all, sorry for my absence. 

Yes, the boundaries were put in place as a compromise. Initially someone had put in boundaries to reflect the Creality advertised XY area of 220x220. Many of us wanted it gone since the technical specs and build allowed access to the full 235x235. But others were concerned with safety, especially for new users. I proposed we then only prevent travel at the front and back where the clips are so we can at least recover the sides. This was acceptable so I went ahead and created the necessary changes which were then approved and accepted.

Personally I just go into the file and take them right back out after installing Cura.

However, if indeed the Ender 3 Pro does not come with clips at all then I would agree with the suggestion to add a profile distinguishing the "standard" from the "pro." In the last discussion, Ultimaker developers wanted original source proof, so I imagine that needs to be provided here too. Assuming that is provided, @Ghostkeeper, would adding a "pro" profile be acceptable to you? If so, I'm happy to do it since I already have the structure in my repository.

@mathiasvr if you need to know how to get rid of them on your installation, PM me and I'll help you out.

murphydan, 2020-04-08:
I completely agree with changing the Ender 3 profile to be 235mm instead of 220mm. I stumbled upon the that huge thread linked to earlier in this issue and was excited to see the print area changed to 235.235 but think the boundaries should just be removed completely. If someone is printing with clips attached to their bed, they should need to account for that when placing their model.

I'll gladly create the PR to "fix" this (keeping the larger 235mm build volume and removing the boundaries) if that is what is needed but I'm guessing there is going to be some debate on this one.

Regardless, are there instructions somewhere I can follow to change my own installation to my liking? Thanks!

murphydan, 2020-04-08:
Ok, I finally found it on a mac. I assumed it would be in the `{~}/Library/Application Support/cura/4.5/definitions` directory but that is empty. The definitions are all actually listed here...

`/System/Volumes/Data/Applications/Ultimaker Cura.app/Contents/Resources/resources/definitions/`

I was able to remove the `machine_disallowed_areas` and keep the size of 235x235. I'm happy! I'll test it out when my next print finishes, but it appears as though I am no longer going to print everything off-centered due to the 220x220 size that was there before.

I'm willing to help in whatever way needed when we figure out the way forward on this one.

Liger0, 2020-04-08:
> 
> 
>  Plus, you could still print to the edge with clips, just not right where the clips are.

Not really. The clips are not accounted for travel moves, where you can still hit them even not placing the model around them.

murphydan, 2020-04-08:
Yeah, I didn't think of travel, good point. Still, folks are using all sorts of ways to hold down the build plate so it is tough to pick a great default, especially for Ender 3 Pro users. I'm leaning towards a new Ender 3 Pro config that keeps the 235x235 with no `machine_disallowed_areas`. There are a couple of the other "Pro" models in the Creality list so there is already a precedence.

What would be awesome is if this was in the UI somehow like some of the other settings (machine_head_with_fans_polygon, gantry_height, machine_start_gcode, etc.). Then if someone is using big front clips, corner clips, has a cold left side of the bed, etc. they could just set the non printable area themselves.

There could even be an argument that the left .5mm should be disallowed by default since the default `machine_start_gcode` for the Ender 3 puts a line up and down from .1mm to .4mm. 😬

Ghostkeeper, 2020-04-08:
> Ok, I finally found it on a mac. I assumed it would be in the `{~}/Library/Application Support/cura/4.5/definitions` directory but that is empty. The definitions are all actually listed here...
> 
> `/System/Volumes/Data/Applications/Ultimaker Cura.app/Contents/Resources/resources/definitions/`

Cura should read definition files from both of these directories. We added the one in Application Support to make it easier for people to add definitions manually, since the other one in the .app file is write-protected and signed so your computer shouldn't allow you to run Cura after that is modified.

Ghostkeeper, 2020-04-08:
If the Pro model doesn't have those clips but does have a fully useable bed with the default firmware, then I'm all for adding that printer definition for the Ender 3 Pro that specialises the build volume and removes the disallowed areas.

> What would be awesome is if this was in the UI somehow like some of the other settings (machine_head_with_fans_polygon, gantry_height, machine_start_gcode, etc.). Then if someone is using big front [...]

The "Printer Settings" plug-in from the Marketplace does this precisely. It's a useful add-on to iterate quickly if you're designing your own definition files.

kazooless, 2020-04-08:
@Ghostkeeper I'll create a new definition file for the Pro and submit a PR then (unless someone else beats me to it)

kazooless, 2020-04-11:
Okay, the pull request is submitted.

https://github.com/Ultimaker/Cura/pull/7466

Ghostkeeper, 2020-04-14:
Thanks! Cura 4.7 will now have a separate printer definition for the Creality Ender 3 Pro that doesn't have disallowed areas for the clips any more.

closed 2020-04-14 by Ghostkeeper

murphydan, 2020-04-15:
> The "Printer Settings" plug-in from the Marketplace does this precisely. It's a useful add-on to iterate quickly if you're designing your own definition files.

@Ghostkeeper - Thanks for the tip about the plug-in. What I was talking about specifically is the `machine_disallowed_areas` parameter in the config and it doesn't look like it is in the "Printer Settings" plug-in.

I have an Ender 3 Pro so I can print edge to edge with the magnetic bed. But, I actually just plopped on a polypropylene plate on top to test printing on it and am holding it in place with clips. Because I now know where these config files are, I was able to change my printer's config to reflect the addition of the clips in the non-printable area. But, I'm looking into thermal pads now to try and get rid of the clips (and I'll then adjust my printable area again in the config files).

So the point I'm making is it would be nice to have the `machine_disallowed_areas` parameter in the UI somehow and it seems like the only way to do that is in the config files. Would this be a feature request created as an issue? New to this repo and I don't see feature request info listed in the readme. Thanks!

Ben-Boardman, 2020-04-17:
> Removing these lines fixes the issue:
> ` [[-117.5, 117.5], [-117.5, 108], [117.5, 108], [117.5, 117.5]], [[-117.5, -108], [-117.5, -117.5], [117.5, -117.5], [117.5, -108]]`

I also had the same issue. Removed these lines and the profile is now using the whole bed. Many thanks!

Ghostkeeper, 2020-04-21:
Oh, I'd think that this Machine Settings thing would make that visible but maybe it doesn't because there is no UI element for lists of coordinates really.

renlewis, 2020-04-22:
I tried this change and it worked when there was no object on the bed.  As soon as I loaded an .stl file the disallowed areas showed back up.  Any thoughts?

kazooless, 2020-04-22:
It is probably a raft, brim, or skirt.

_____
Jeff Kazules

Ghostkeeper, 2020-04-28:
There is also a border around the printer that depends on which extruders are in use and the settings on those extruders. The bed adhesion extruder is always in use except if the adhesion type is None. So probably you had adhesion set to None and then an extruder was in use by printing your model.

There are dozens of settings that influence this border and the disallowed areas. They are listed here:
https://github.com/Ultimaker/Cura/blob/68e8261c2124415814b89809fc8c914e2e4f3bde/cura/BuildVolume.py#L1135-L1144