Ultimaker/Cura #7253: Cura [4.5.0] Incorrect display of program screen with Kubuntu 19.10

Gleis1234, 2020-03-10:
4.5.0
**Application version**
(The version of the application this issue occurs with.)
Kubuntu 19.10
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
doesn't matterr
**Printer**
(Which printer was selected in Cura?)
Start Cura
**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

Screenshot attached
**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
Not all Information on screen is displayed.
**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

This happens after upgrading Kubuntu from 19.04 to 19.10
Cura 4.4.1 has the same problem.
[image]

**Additional information**
(Extra information relevant to the issue.)

Gleis1234, 2020-03-10:
Adding Logfile
[cura.log](https://github.com/Ultimaker/Cura/files/4311277/cura.log)

fieldOfView, 2020-03-10:
See https://github.com/Ultimaker/Cura/issues/5296

Gleis1234, 2020-03-11:
Thank you for the quick help. The information solved my problem.

closed 2020-03-11 by Gleis1234