Ultimaker/Cura #7254: [4.5] Initial purge extrusion picks up its own debris

top-gun, 2020-03-10:
**Cura 4.5 (pertains to 4.4.1, too)**

**Windows 10 Pro, 1904**

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Pre-heat nozzle and bed for 10 minutes
2. Start a print on PLA, 205° and 60° 

**Actual results**
At the starting point, the print head will often release a few centimeters of extruded plastic. This will usually fall off at the beginning of the "initial purge run" where the head runs along the left side of the printing bed.

The head runs along the y-axis, then it returns to the exact starting position, before it lifts off for skirt/brim-printing. When it returns to the starting point, about 50% of the times it will pick up the old debris and carry it over to the skirt-position. This contaminates the actual printing area with a string of extruded cold plastic.

**Expected results**
At the end of the purge run, the print head should stop 1-2cm before the starting position and lift off, so it avoids the debris.

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
This is a sample gcode-file from my last print:

`;FLAVOR:Marlin
M117 Time Left 9h11m52s
;Filament used: 29.6744m
;Layer height: 0.2
;MINX:19.775
;MINY:38.287
;MINZ:0.2
;MAXX:200.225
;MAXY:202.225
;MAXZ:19.88
;POSTPROCESSED
;Generated with Cura_SteamEngine 4.5.0
M140 S60
M105
M190 S60
M104 S205
M105
M109 S205
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
G28 ; Home all axes
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 **_Y20_** Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
`

My suggestion: Don't return to "Y20", use "Y40" instead. This will usually stop before the head reaches the old extruded material.

mahtDFR, 2020-03-10:
For your situation, you can edit the start gcode in the machine settings menu. This code will be inserted prior to the job start.

[image]

top-gun, 2020-03-10:
Thank you for your reply, I finally found out how to make the code visible by manually changing the window size :-)

Anyway: The code that is delivered in Cura leaves room for perfection. I understand the purge extrusion is meant to get filament flowing and clean the nozzle. By ending the drive right were you started (ok, 0.3mm to the right) the hot nozzle drives straight into the waste that it just dumped on the left side. Ending the purge extrusion 2cm earlier would improve this situation.

Ghostkeeper, 2020-03-13:
We can't make changes to the start g-code for your printer ourselves, since we don't have your printer in the office to test with. We rely on the community to make these changes for us. We'd like multiple people to test this since it may also just be a problem specific to your printer.

If you guys tested this and agree that the change makes it better, we can merge it in our source code (preferably via a pull request).

closed 2020-03-13 by Ghostkeeper