Ultimaker/Cura #7255: [4.5] Max temperatur only 365°C

Prio94, 2020-03-10:
**Platform**
Windows10
**Printer**
Creatbot F430 (for PEEK)

**Reproduction steps**

    Load Stl
    Try to slice

**Actual results**
Not slicing...
Field for the temperatur becomes red (not orange).
This didn´t happen in [4.4] (only orange).

**Expected results**
I want to print with high temperatures, above 400°C.

Goodfeat, 2020-03-11:
If you want to print at higher temperatures, you need to edit the file fdmprinter.def.json or add your own configuration for your printer, where this restriction will be removed

[image]

This workaround will work on all current versions of the program.

fieldOfView, 2020-03-11:
> or add your own configuration for your printer

This is the best option. Don't edit fdmprinter.def.json.

fieldOfView, 2020-03-11:
It is a good question why there is an arbitrary maximum value for temperatures. In other places (temperatures, dimensions), Cura makes a point of not placing a hard limit on values other than limits of physics such as the speed of light.

nallath, 2020-03-11:
For quite a few materials, it can get dangerous (and even lethal) if you heat them higher. So that's why these do have a max temp.
Furthermore, marlin also has these values as the max per default (and the vast majority of printers that are based on fdm_printer.def.json use that). 

The best solution, in this case, is to add a custom definition for your printer that increases the max temperature.

closed 2020-03-11 by nallath