Ultimaker/Cura #7256: Layer Dependent Values

bjoern79de, 2020-03-10:
One thing I miss in all available slicers is the ability to use formulas instead of hard values for various settings.
I'd like set i.e. ` retract-length = 1.5 + {{layer-number}} * 0.05` to be able to print only one object with a range of values (in a fine resolution) and measure out at which heigh the best results appears. Currently I have to print a large amount of parts (or have to make a huge bunch of definitions manually) to find out the best values for retraction-length/speed, general speed, temperature, "pressure-advance", extrusion-multiplier, .....

nallath, 2020-03-11:
You could try and use the "tweak at z" postprocessing script.

bjoern79de, 2020-03-11:
As far as I know it can just manipulate values at specific points and not continously.

nallath, 2020-03-11:
Changing the slicer in such a way that this is possible is simply not feasible. You could have a look at services like https://3doptimizer.com/.

Ghostkeeper, 2020-03-13:
Duplicate of https://github.com/Ultimaker/Cura/issues/4235.

closed 2020-03-13 by Ghostkeeper