Ultimaker/Cura #7258: Turn Off Helpers

hatrickseanmikal, 2020-03-10:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-03-10:
We expect you to fill in the template. An issue is never just the title; we need more information.

Ghostkeeper, 2020-03-13:
AH, I SEE YOU ALSO INSTALLED A CRUISE CONTROL BUTTON ON YOUR KEYBOARD.

> I EXPECT TO BE PROVIDED WITH A WAY TO TURN OFF OR REMOVE THE " HELPER SUPPORTS" AS THEY ARE NOT NEEDED

INDEED, YOU ARE PROVIDED WITH A WAY TO TURN OFF SUPPORTS:
[image]

closed 2020-03-13 by Ghostkeeper