Ultimaker/Cura #7261: Support blockers vs. Custom supports

tameddragon66, 2020-03-11:
In an effort to reduce the amount of custom support generated where it was not needed, I tried placing support erasers in strategic areas. When sliced, the entire rectangular area generates support and the erasers had no effect on the custom support area. Separately, both work fine.

**Application version**
Cura 4.5.0

**Platform**
Windows 10

**Printer**
Creality Ender 3

**Reproduction steps**
1. Placed a custom support 
2. Placed a support eraser
3. Sliced model

**Screenshot(s)**

[image]
[image]

**Actual results**
All supports indicated were generated

**Expected results**
No supports generated in area marked by support eraser

**Project file**

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-03-11:
This is expected behavior. Support Meshes overrule Support Blockers.

Support Blockers aren't actually subtracted from supports. Support Blockers mark parts of a mesh which would normally be supported to be ignored by the automatic support generation algorithm. Custom support blockers do no use those algorithms, so Support Blockers have no influence on them.

tameddragon66, 2020-03-12:
fieldofView,
   Yes, it is apparent that is how it functions. However, would it not be more useful to have the eraser function in the manner that I was attempting to effect? Thanks for your response.

closed 2020-03-12 by tameddragon66

Ghostkeeper, 2020-03-17:
Originally these meshes were called "anti-overhang meshes" because they block generating support where there is overhang in the mesh, rather than erasing support itself. It was renamed because "anti-overhang mesh" is hard to understand and to a user, blocking support is what it really appears to do. I agree though that it would be more useful to block the support itself too. We don't really have a function for that at the moment.