Ultimaker/Cura #7262: Unable to do a clean install

rromano158, 2020-03-11:
I am using Ultimaker Cura 4.5. I am having some issues with corrupted profiles, and have tried twice to do a clean install, but every time I have re-installed the program it still shows up with my printer name and previous profiles. When I click on "Help" then "Show Configuration Folder", nothing happens, no other window pops up to show the configuration folder. I have even went into regedit to delete anything with the name of Cura, rebooted, re-installed the program and still shows my printer and previous profiles. How can I eradicate this program from my computer and start with a clean install?

fieldOfView, 2020-03-11:
Cura does not store configuration data in the registry. Configuration data is stored in the "appdata folder". You can start with a fresh installation of Cura like this:

Press the windows key on your keyboard, release it and start typing *exactly* the following:
`%APPDATA%cura\4.5` followed by enter.

Remove all the files from that folder, but leave the empty folder in place. Then restart Cura.

I think you have installed Cura using an MSI file. The normal EXE installer has an uninstaller that has the option to remove the configuration data. You could also download the EXE installer, install Cura 4.5 over your current Cura 4.5, and then use the uninstaller. It will ask you if you want to remove your configuration data.

Ghostkeeper, 2020-03-13:
The MSI actually does put stuff in the registry. Not actual Cura configuration that Cura is using. Just stuff that Windows and the uninstaller uses.

See also https://github.com/Ultimaker/Cura/issues/7183 for a different bug that would be solved by having the MSI installer have an option to erase these config files.

closed 2024-12-07 by github-actions[bot]