Ultimaker/Cura #7263: Cura 4.5.0 ignores Skin Overlap Percentage values causing gaps on skin surfaces

sekbowee, 2020-03-11:
**Application version**
Latest Cura 4.5.0

**Platform**
Windows 10 Pro 64-bit
GTX 980 Ti

**Printer**
Creality Ender-3 (Selected printer in cura)

**Reproduction steps**
1. Changing values of Skin Overlap Percentage to 5 - 50 %
2. Nothing changed on the quality on the skin (no differences, gaps present)

**Screenshot(s)**
[image]

**Expected results**
1. Skin overflow onto perimeter/walls
2. No more gaps between skin and wall.

**Actual results**
[image]

**Profile**
[Creality Ender-3 PLA.zip](https://github.com/Ultimaker/Cura/files/4316633/Creality.Ender-3.PLA.zip)

Ghostkeeper, 2020-03-13:
I'm not able to reproduce your bug. With the overlap set to 20 micrometres the overlap is pretty much invisible so I upped it to 0.8mm, and you can clearly see that the skin lines are overlapping with the walls:
[image]

Perhaps the gaps are caused by other things. Tried upping the print temperature a little bit?

So far I don't see any bug here in how Cura behaves.

sekbowee, 2020-03-14:
I just found out that if u had Top Surface Skin Layers set to > 0 , the top skin will ignore the skin overlap setting

[image]

If Top Surface Skin Layers set to 0, top skin will follow the skin overlap setting

[image]

Ghostkeeper, 2020-03-18:
Ah, yes. That's a known issue, sorry. I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/5758.

closed 2020-03-18 by Ghostkeeper