Ultimaker/Cura #7265: [4.5.0] Holes in wall the same thickness as nozzle and top layers missing

Carxos, 2020-03-11:
**Application version**
4.5.0

**Platform**
Windows 10 Pro
nVidia GeForce GTX960

**Printer**
Ultimaker 3

**Reproduction steps**
Load the mesh with one wall that is the width of the nozzle
Use Profile Sprint
Slice

**Screenshot(s)**
[image]

**Actual results**
Hole appears in preview and the top layers are missing

**Expected results**
No holes in preview and all layers should be present

**Project file**
[Transport.zip](https://github.com/Ultimaker/Cura/files/4318404/Transport.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4318419/cura.log)

**Additional information**

Vandrasc, 2020-03-16:
Hi @Carxos , I also checked in previous versions and there is the same behavior, when printing with 0.8 printcore the top layers are missing. Let me check with our developers and will get to you.

rburema, 2020-03-17:
I can confirm that this is in current master. I've also inspected the model and everything seems to be in order there.

It doesn't happen if the width of the walls is set to less than the model width... something like 0.74 for the missing top layers, and the missing patch in the curled section will appear from 0.72 onward).

I think it's good to make a ticket out of this, as a straight patch of the 'hook section' is also missing, and it only comes back at 0.72mm wall width. So there _could_ be more going on than just rounding errors & edge cases.

Since it's relatively easy to work around though, and we have a lot of bugs on the backlog a.t.m., I don't see this getting fixed that soon. (Especially since we plan to look at these kinds of issues in a more general manner some time this year.)

Reference to UM-internal ticket (as opposed to GH issue): CURA-7303

Ghostkeeper, 2020-03-17:
This is indeed a bug, and it's in `simplify`. The bug is caused by the shape of the triangles there. The triangle that forms this face ends in a thin point at the top-left corner (as seen from the above screenshot). There the line segment introduced by this triangle is then small enough to be considered for removal by the simplification (i.e. smaller than Maximum Resolution). When it gets removed, the deviation from the original outline causes the wall to be slightly thinner than the line width, making it disappear completely on those layers.

This is one of the things that would probably be fixed by this PR: https://github.com/Ultimaker/CuraEngine/pull/1214

For now, a workaround is to reduce the Maximum Resolution setting slightly (to get a higher resolution curve there). 0.3mm seems to work for your case.

Carxos, 2020-03-17:
I solved it by doubling the wall thickness to two nozzle widths. This did not increase printing time but increased filament usage. The model is only made in 13 copies so not crucial to limit filament usage.

nallath, 2020-03-30:
I can confirm that the hole in the wall issue is fixed by https://github.com/Ultimaker/CuraEngine/pull/1214

GregValiant, 2024-11-14:
Has it remained fixed and can this be closed?

Carxos, 2024-11-15:
Printed the same model without problem

closed 2024-11-15 by Carxos