Ultimaker/Cura #7267: [4.5.0] doesn't update Material Temps when filament is changed.

PuterPro, 2020-03-11:
If I change Materials, the first four materials temps do NOT update to the new material. Build Plate DOES.
You get the Dialog to confirm changes the FIRST time but further changes you don't, if you haven't changed some OTHER parameter like speed or # of layers. If you change something, you DO get the Dialog box but it does NOT flag the Material Temp changes, and leaves them alone while correctly changing bed temp.

**Application version**
4.5.0

**Platform**
Windows 10; nVidia RTX2070 Super, latest updates for both.
**Printer**
Creality CR-10S

**Reproduction steps**
1. Change Materiel from a PLA to a PC (for example)
2. Accept changes in Dialog. Bed temp and other changes update, Filament temps do not.

**Screenshot(s)**

**Actual results**
Explained already. Now, I'd like to add here that this MAY be related to bug report #7250 https://github.com/Ultimaker/Cura/issues/7250

**Expected results**
Temps should update.
**Project file**
Any.

**Log file**
Log & Config:
[cura_4.5.0.zip](https://github.com/Ultimaker/Cura/files/4321176/cura_4.5.0.zip)

PuterPro, 2020-03-13:
It seems I didn't upload the config file right, here it is:
[4.5.0_config.zip](https://github.com/Ultimaker/Cura/files/4331341/4.5.0_config.zip)

PuterPro, 2020-03-14:
Update: I was also experiencing #7250 as I mentioned. Today, chasing that, I did a complete wipe of ALL Cura installs (I had 4.3 & 4.5 loaded), after I Exported all Profiles and Materials.

I'm a PC tech so I did a thorough scrub of everything, ALL files, folders (hidden etc) and removed ALL Registry entries for Cura & Ultimaker. FULL uninstall.

I then loaded the latest 4.5 EXE (not msi). Went flawlessly. I then Imported all my Profiles and Materials. Still fine.
Closed, and reopened Cura, and loaded an STL. Still fine. It was set to the Default PLA (Generic) @ 205°C.
When I chose a new Material (Polymaker PC-Max) , and **it did NOT update the 4 Hotend temps**, as reported in this bug report.

I then changed a PROFILE to a custom one (Fine .12 PLA+ Raft,coasting & Adaptive) and two things happened;
1. The temps changed to the proper Material Temps. (260°C)
2. The STANDARD Profiles VANISHED, as reported in #7250 .

BUT - When I changed to one of the CHEP profiles [CHEPV4.5_012(Best) ], THE TEMPS WENT BACK TO PLA (205°C). Toggling to MY profile (Fine .12 PLA+ Raft ...) will switch the temps to correct 260°C.

NOTHING I do, however, will bring back the Standard Cura Profiles, they're GONE.

So, the bugs are real, and apparently related.

Here are the Profiles and Material, and STL files I loaded that triggered the behavior.

[[4.5.0] Temp & Profile Bug.zip](https://github.com/Ultimaker/Cura/files/4333364/4.5.0.Temp.Profile.Bug.zip)

Ghostkeeper, 2020-03-17:
This is because Chep's profiles override the printing temperature:

[image]

As long as that profile is selected, the printing temperature should be 205 degrees, regardless of which material is used.

closed 2020-03-17 by Ghostkeeper

PuterPro, 2020-03-18:
I'll be sure to inform him about this!
 Thx, I never opened the profiles in Notepad++, it was my first time using an outside source for a profile. Lesson learned, I'll inspect profiles manually next time.
SO sorry for taking your time up!!

Ghostkeeper, 2020-03-18:
`.curaprofile` files are just zip archives, so you could rename them (if your operating system requires) and inspect their contents with like 7zip or something :smiley: Cura should also show the overrides in its profile manager, but this is sometimes confusing with multi-extrusion printers and such.