Ultimaker/Cura #7270: Cura Parameter for Min / Max Print Coordinates?

CCS86, 2020-03-12:
Hey guys, I was hoping that Cura could pass parameters for the minimum and maximum x and y values used in a given print.

My thought is that you could push these values forward into mesh leveling bounds and quickly probe the area of bed being used without waste.

fieldOfView, 2020-03-12:
> I was hoping that Cura could pass parameters 

Please elaborate: pass to where?

CCS86, 2020-03-13:
Sorry, I should have been more specific:

Pass the values via parameter into the start gcode, to be used like these:

M104 S{material_print_temperature}
M140 S{material_bed_temperature}
M109 S{material_print_temperature} 
M190 S{material_print_temperature}

You could then form a probing mesh command like this:

M557 X{x_min}:{x_max} {y_min}:{y_max} S20         ; define mesh grid

fieldOfView, 2020-03-13:
The problem is that the {replacement patterns} are filled in before all geometry and settings are sent to CuraEngine for slicing, and Cura does not *really* know the extends of the print (including any skirt, brim, prime tower, etc) until after Cura is slicing. So there are two options here:

* Cura adds guestimates, which might be not 100% accurate
* This is handled after slicing is done, using a gcode postprocessing script

fieldOfView, 2020-03-13:
Do you expect Cura to include the prime tower in the min/max coordinates? Why/why not?

ChrisTerBeke, 2020-03-13:
The S5 and S3 have mesh bed leveling in the firmware indeed, which uses the footprint of the G-code's 1st layer to determine which part of the bed needs leveling. This would only be a 3rd party printer feature in that case.

CCS86, 2020-03-13:
> The problem is that the {replacement patterns} are filled in before all geometry and settings are sent to CuraEngine for slicing, and Cura does not _really_ know the extends of the print (including [...]

Honestly, even close would be "good enough". If you just took min/max for X and Y where the model touches the build plate, then added a "buffer zone" to expand that a bit, you would be pretty well covered.

> Do you expect Cura to include the prime tower in the min/max coordinates? Why/why not?

This I am not sure about, since I never use prime towers. 

> The S5 and S3 have mesh bed leveling in the firmware indeed, which uses the footprint of the G-code's 1st layer to determine which part of the bed needs leveling. This would only be a 3rd party printer feature in that case.

Does Marlin really read through the entirety of the gcode to find min/max of X and Y, or is Cura already passing these?

I wouldn't say "only 3rd party". Previous versions of Ultimaker would benefit as well. I use a UMO.

fieldOfView, 2020-03-13:
> This I am not sure about, since I never use prime towers.

But there's the thing: Cura has to take that into account, because there are people that do use prime towers that need good adhesion too.

> Does Marlin really read through the entirety of the gcode to find min/max of X and Y,

The firmware for the S5 and S3 is much more than Marlin; there's a whole linux board in those printers that do a lot of handling of specific parts of the gcode, including tool switches and bed leveling. But these printers also use a different "flavor" of gcode, named "griffin", which includes the following header:

```
;START_OF_HEADER
;HEADER_VERSION:0.1
;FLAVOR:Griffin
;GENERATOR.NAME:Cura_SteamEngine
;GENERATOR.VERSION:4.5.0-beta
;GENERATOR.BUILD_DATE:2020-02-11
;TARGET_MACHINE.NAME:Ultimaker S5
;EXTRUDER_TRAIN.0.INITIAL_TEMPERATURE:205
;EXTRUDER_TRAIN.0.MATERIAL.VOLUME_USED:936
;EXTRUDER_TRAIN.0.MATERIAL.GUID:506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9
;EXTRUDER_TRAIN.0.NOZZLE.DIAMETER:0.4
;EXTRUDER_TRAIN.0.NOZZLE.NAME:AA 0.4
;BUILD_PLATE.TYPE:glass
;BUILD_PLATE.INITIAL_TEMPERATURE:60
;BUILD_VOLUME.TEMPERATURE:28
;PRINT.TIME:432
;PRINT.GROUPS:1
;PRINT.SIZE.MIN.X:152.665
;PRINT.SIZE.MIN.Y:107.553
;PRINT.SIZE.MIN.Z:0.2
;PRINT.SIZE.MAX.X:177.333
;PRINT.SIZE.MAX.Y:132.447
;PRINT.SIZE.MAX.Z:9.8
;END_OF_HEADER
```
The print size min/max is parsed from that header for the leveling procedure.

> Previous versions of Ultimaker would benefit as well. I use a UMO.

Though the UMO was extremely moddable, it did not come with a bed level probe. If you modified your UMO to have one, Ultimaker no longer really considers it to be an Ultimaker Original. I do get your point though, but it is relevant for an extremely small portion of Ultimaker printer users only.

CCS86, 2020-03-15:
So, it appears that Cura is already calculating this, as my header contains the values. 

However, it is not passing them as a parameter of the same name, as when I try to reference them, values do not populate:

[image]

Any ideas?

fieldOfView, 2020-03-15:
> So, it appears that Cura is already calculating this, as my header contains the values.

Yes. But like I said before, this is calculated by CuraEngine *after* Cura has already replaced the replacement values for the start gcode. So it cannot be handled by the same code that does the other pattern replacement.

So, either this has to be handled by a postprocessing script, or Cura would have to calculate the min/max positions before sending data to the engine. It is not a question of simply assigning what you see in the gcode to a replacement pattern you can use.

CCS86, 2020-03-16:
> Yes. But like I said before, this is calculated by CuraEngine _after_ Cura has already replaced the replacement values for the start gcode. So it cannot be handled by the same code that does the [...]

Is that order of operations just incidental, or is there a reason it must generate the header code before assigning values?

fieldOfView, 2020-03-16:
There's a reason. But that is irrelevant; a non-trivial change needs to be made to support your case (either a postprocessing script needs to be written or some code needs to be written to add the replacement parameters you request). Because this is only going to be useful for 3rd party printers and a very small number of fairly heavily modded older Ultimaker models, chances are small that Ultimaker employees are going to be allotted time to do this.

In other words: unless a community member like you or me is going to implement this, it is not likely going to happen.

fieldOfView, 2020-03-16:
Something I just realized is that the min/max values you see in the gcode header are for the whole print. If you have a model that is narrow on the bottom, but wide on top, those values will not do you much good. You really want the min/max values for the bottom layer (including adhesion helpers and the purge tower if one is enabled). This can only be reliably done after slicing, so it would need to be done with a postprocessing script (or in CuraEngine, but I cannot help you with that).

ChrisTerBeke, 2020-03-16:
> Something I just realized is that the min/max values you see in the gcode header are for the whole print. If you have a model that is narrow on the bottom, but wide on top, those values will not do [...]

You're right. And the S3/S5 multi-point leveling only look at the min/max coords of the 1st layer.

CCS86, 2020-03-16:
I understand.

My hope was that either it already existed and I just needed to know the formal parameter name, or that it could be made available with very little effort. I understand not wanting to spend the developer time on something not applicable to many.

Ghostkeeper, 2020-03-17:
I couldn't have worded it any better than fieldOfView did here (with Chris' technical corrections on how the S3 and S5 implement this). For that reason, we'll mark this issue as deferred, as in "technically Cura could support this, but we won't get any priority on this unless it's necessary for Ultimaker".

The difficulty with making a guesstimate is in support. If you use tree support or conical support with a negative angle, who knows where support will end up?

closed 2023-07-13 by github-actions[bot]