Ultimaker/Cura #7272: Enhanced Retraction Options

neildmd, 2020-03-13:
I feel that a combination of retract while combing and z-hop is the current best route to eliminate stringing for most common prints. What we truly need is the ability to retract and hop after the wipe/comb motion has occurred. This should eliminate stringing for parts where z-hop is required due to thing features curling up around the edges.

Another idea would be to incorporate a ramp option for z-hop where we can set a horizontal distance over which the hop takes place and have the nozzle move towards infill just like combing. 

Yet another, if z-hop could have a timing configuration option that wcould also be very useful. e.g. Wait X milli-seconds after retract before hopping. This ensures any strings or blobs get deposited within infill before the hop is made. Detection and avoidance of small features during travel would be practical in this case as well.

Just some ideas, feel free to expand or recommend your own solution if you think there's a better one.

Ghostkeeper, 2020-03-17:
For us it's usually easier to separate out different issues. Once one of your features is implemented, we'd close the issue and forget about the rest.

Also, not sure if you're really getting the terminology right here. Combing means the practice of trying to avoid the walls when making a travel move.

> I feel that a combination of retract while combing and z-hop is the current best route to eliminate stringing for most common prints.

The logic is that if you're making a Z hop, you'll be moving over the top of the walls without hitting those walls. This feature request is a duplicate of https://github.com/Ultimaker/Cura/issues/5060.

> Another idea would be to incorporate a ramp option for z-hop where we can set a horizontal distance over which the hop takes place and have the nozzle move towards infill just like combing.

I think it could save a bit of time due to accelerations, but the change will be very minimal. This feature request is a duplicate of https://github.com/Ultimaker/Cura/issues/2761.

> Yet another, if z-hop could have a timing configuration option that wcould also be very useful. e.g. Wait X milli-seconds after retract before hopping. This ensures any strings or blobs get [...]

Is this more effective than reducing the retraction speed?

neildmd, 2020-03-17:
Ok understood, I've solidified my feature request. 

I do perfectly understand how combing works and this is why I'm suggesting it. As an effect of avoiding walls it eliminates stringing by depositing the material inside the print. 

This alone is not enough though because some parts require z-hop as explained above and this reintroduces stringing due to the way it's implemented. This feature is widely applicable to PLA based and flexible filaments on a broad spectrum of FDM printers.