Ultimaker/Cura #7274: No filiament retraction when printing Raft

zelosos, 2020-03-13:
**Application version**
4.5.0

**Platform**
Linux Mint 19.3, 64Bit with Cinnamon

**Printer**
MonopriceSelect Mini V1 

**Reproduction steps**
1. Import Model
2. Use Slice Settings as you want
3. Use **Raft** for adhesion method
4. Print the part

**Screenshot(s)**
sorry no images

**Actual results**
When printing the part there is **no retraction** happening when printing the **Raft**.
This results in Strings in the Raft and the next layer gets bumpy.
Retraction when printing the actual part is working fine.

**Expected results**
When moving while printing the Raft retract the filiament before moving to the next location. Then continue printing.

**Project file**
Happens on every object. So choose some you like.

**Log file**
This is not neccessary. You should be able to replicate the Bug without Log.

**Additional information**
Use the Retraction code from normal printing, when printing the Raft.

rburema, 2020-03-17:
Have you tried 'Retract at Layer Change'?

zelosos, 2020-03-18:
It did not happen at Layer Change. It happen while printing one layer. When moving to a other location at the same layer.

rburema, 2020-03-18:
You could try setting Max Comb Distance Without Retract to a low value (other than 0, which disables the setting).

Ghostkeeper, 2020-03-18:
Or maybe disabling combing altogether. Is a bit of stringing in the raft all that bad though?

zelosos, 2020-03-18:
It does not destroy the print, but makes the first layer sometimes rough.
Disabling combining does only affect the raft?

But I mean the code for retraction in printing mode exists, why not use them in the raft as well?

Ghostkeeper, 2020-03-19:
No you can't disable/enable combing for the raft specifically.

Ghostkeeper, 2020-03-19:
It does retract in the raft, in the same way as when it would retract in the model. Only the raft is not made out of skin or infill, so "Skin" or "Infill Only" would be the same as "everywhere". Also the raft is often made out of one piece due to its large offset width, so it doesn't need to travel to different pieces often (which would normally incur a retraction).