Ultimaker/Cura #7275: [4.5] slicer gcode has excessive 80mm retraction at end of print for Anet printers

pjossman, 2020-03-13:
**Application version**
4.5.0

**Platform**
Windows 10 Home v 1909

**Printer**
Anet A2 PLUS in my case but applies to all Anet printers

**Reproduction steps**
1. Slice a very small model
2. a)Print the model and at the end the filament is retracted 80mm up the bowden extruder tube. 
OR
2 b) look at the final extrusion instruction in the generated gcode that indicates an 80 mm retraction "G1 E-80 F2000"

**Screenshot(s)**

**Actual results**

**Expected results**

**Project file**

**Log file**

**Additional information**
The problem is actually in the code in the resource file: 
Utilmaker Cura 4.5\resources\definitions\anet3d_a2_plus.def.json. 

specifically the instruction in section below "G1 E-80 F2000" causing the excessive 80mm retraction when the print ends.
"machine_end_gcode": {
            "default_value": "M104 S0\nM140 S0\nG92 E80\nG1 E-80 F2000\nG28 X0 Y0\nM84"

The earlier versions of Cura retracted the filament about 5 or 6 mm. I changed the 4.5.0 resource file myself to fix this problem but wanted to report it. Otherwise you have to extrude 80 mm of filament to get it into the nozzle for the next print.

nallath, 2020-03-16:
Well, this is how it was contributed to us. Maybe @springtiger, who contributed them, can say something about this?

JanGerene, 2020-03-24:
Same problem for Anet A6.  The filament is retracted too much at the end of the print.

pjossman, 2020-03-24:
In case people want to fix it themselves, I changed the values in the definition file as per the **bold values** in the string below. This changes the retraction to 10mm which works fine. I would like to fix it in the source files in the distribution itself but am not familiar with the procedures for doing so. As noted above this problem is in the definition files for all Anet printers.

The specific values changed are:
E80  -> E0
E-80 -> E-10

"machine_end_gcode": {
            "default_value": "M104 S0\nM140 S0\nG92 **E0**\nG1 **E-10** F2000\nG28 X0 Y0\nM84"
        }

ryan77627, 2020-03-26:
Yeah, I can confirm my Anet A8 had this as well on a clean install of the latest version of Cura. I was pulling my hair out as to why I had to keep reloading the filament.

Ghostkeeper, 2020-03-26:
Since multiple people confirmed this problem I've applied the proposed change for the next version for all Anet3D printers.

closed 2020-03-26 by Ghostkeeper

springtiger, 2020-03-29:
I'm sorry that I didn't reply in time due to the Gmail email access problem. I will try to fix this problem in the next version

commit 92691f68 (2020-03-26): Reduce retraction at the end of all Anet g-code

Multiple people have found this to work better.

Fixes #7275.