Ultimaker/Cura #7276: [4.5.0] eSUN PLA+ Not Supported

htrabbit, 2020-03-14:
when i select esun pla+ filament, it shows as not suported.
i am using the built inn ender 5 profile. i tried to change layer height, width, nozzle etc. 
no change 
-->

**Application version**
4.5.0

**Platform**
Windows 10

**Printer**
Tronxy and/or ender5

**Reproduction steps**
1. Load an stl file
2. change filament to Built in eSUN PLA+ (any color)

**Screenshot(s)**
[image]

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-03-16:
This is as expected; A specific quality profile was not provided for this material. So although you can print with it, it won't have great settings (which is also why it's marked in orange instead of red. It's only warning you that you're on your own, not preventing you from doing it).

closed 2020-03-16 by nallath

diamaunt, 2020-06-08:
What is the point of even having profiles in there when they're just going to say "HAH! you're on your own with this one"?  

I'm new, humor me.

htrabbit, 2020-06-11:
maybe what we need is a way for us to create these profiles, say a small utility and or repository of some kind

diamaunt, 2020-06-11:
Either that, or remove the unsupported materials from the default cura file, If there hadn't been an ESUN PLA+ material listed, there would never have been any question, I'd have just made up my own, but Cura had to dangle a carrot, and then say "Silly wabbit, trix are for kids, go away"

htrabbit, 2020-06-11:
HAHA. But there is some merit to that statement. if not there i would just create my own, but when there......you expect it to work. 
that being said, this is still by far still the best slicer IMHO.

Ghostkeeper, 2020-06-15:
Ultimaker has a utility to guide material engineers to create profiles for Ultimaker printers. It's only distributed to material manufacturers though, and it's limited to Ultimaker printers since it has quite a lot of specifics, such as how the material works in the Ultimaker Material Station. It's part of a programme that also includes making the resulting profiles featured in the Marketplace. That's why there are 214 featured material profiles on the Marketplace, and only one that's not featured :laughing: The one not featured is made by me, by hand.

We can't outright remove a material from Cura since other printers do use and support it.
A printer can say that some materials are incompatible with it and then those won't show up in the list. However we'd rather not do this for profiles that are unknown, because then nobody will ever be able to create profiles for those materials.

Ofloo, 2020-09-24:
The not supported is because it's grouped as PLA+ instead of PLA for some reason cura doesn't support PLA+ material group. Know the problem and how to fix it but not where.

Ghostkeeper, 2020-09-28:
Of course, it will then use the profiles tuned for PLA.
You could also change the material type to ABS, and it'll use the profiles tuned for ABS then.

The problem here is that we don't have any profiles for PLA+ for the printers in question.

Ofloo, 2020-09-28:
What are you talking about the profile says it is PLA+ in the description, it used to work as well but it doesn't anymore. It is an installed lib from market. My own custom profiles used to work like this as well to my surprise they had the same issue.

In 4.5.0 their profile used to color orange but you where able to select a profile. This isn't the case any more in 4.7.x
In any case if you just copy the eSUN profile you can customize it to PLA instead of PLA+ and then the profiles become available.

I assume the eSUN market materials are based on PLA. Since there have never been any PLA+ profiles. Tested it and just prints fine.

htrabbit, 2020-09-30:
my printer prints pla, pla+,petg,pc,,cf-nylon. I mean really i just need to customize the print temp and bed temp, maybe retraction and color setting per filament, not psycho analyze my printer. even if i clone the generic pla profile, as soon as i set it to pla+....poof no printer profile. to blame the filament manufacturer at this point is a cop out. there is an issue here. i just don't have hope it will be resolved.

as i side note ......in Prusa slicer i can create any filament i want, call it anything i want and it will work with any printer i setup

Ghostkeeper, 2020-10-02:
> What are you talking about the profile says it is PLA+ in the description, it used to work as well but it doesn't anymore. [...] In 4.5.0 their profile used to color orange but you where able to select a profile. This isn't the case any more in 4.7.x

It used to place an arbitrary random built-in profile underneath your custom profile if there are no built-in profiles available. That may have been fine if you're lucky, but is very unreliable for most users.

> It is an installed lib from market.

eSUN PLA+ is [not on the Marketplace](https://marketplace.ultimaker.com/app/cura/materials?page=1&search=eSUN) as of this writing. Regardless of its source, there will always be combinations not supported, because no material manufacturer is going to create profiles for all 3D printer models and all of their configurations out there, and no printer manufacturer is going to create profiles for all materials out there.

> even if i clone the generic pla profile, as soon as i set it to pla+....poof no printer profile. to blame the filament manufacturer at this point is a cop out. there is an issue here. i just don't [...]

I think your frustration arises from a misconception that the "material type" is just an arbitrary name you can change to whatever you like. But it's an actual field used by Cura to determine which set of profiles to assign to the material. If you change it to something arbitrary like "Chocolate" then Cura will claim (correctly) that it has no quality profiles to print chocolate with.
This misconception needs to be cleared up, by renaming that field to something more indicative of Cura's behaviour and perhaps by turning it into a combo box to select from existing material types that do have profiles.
As a side note .....in Cura slicer I can create any filament I want, call it anything I want by simply adjusting the field called "Display name". Just don't change the Material Type unless you want to change the set of profiles that the material is compatible with. It's not just for display... The display name is.

diamaunt, 2020-10-02:
What is the point of having built in profiles that are *unsupported*? It just confuses people, and wastes their time.

htrabbit, 2020-10-02:
i think the problem is that if create a profile called "Chocolate" then Cura Should ASK me what quality profiles to print chocolate with.

Ghostkeeper, 2020-10-05:
> What is the point of having built in profiles that are unsupported? It just confuses people, and wastes their time.

There are material+nozzle+printer combinations that are unsupported because there are *no built-in quality profiles* for them. So no, there are no quality profiles that are unsupported. Having a quality profile by definition means that it'll be supported, since it only shows up as not supported if there are no profiles for it.

We could hide such combinations from the user so that you cannot select them. But if you can't select them you can't create any profiles for those combinations yourself either. That would protect users more but also limits them such that they can't create their own material types any more.

> i think the problem is that if create a profile called "Chocolate" then Cura Should ASK me what quality profiles to print chocolate with.

At the risk of repeating myself: This issue only concerns the material type which you manually need to change in one of the field and which alone determines the set of quality profiles. So if you change the Material Type to "Chocolate" and Cura would ask you which set of quality profiles to use, it would change the material type back to whatever you answered to the prompt. I.e. you can't change the material type to anything new then.

diamaunt, 2020-10-06:
> There are material+nozzle+printer combinations that are unsupported because there are no built-in quality profiles for them.

Then, why are they even listed in the default profiles?

like, Esun PLA+... it's listed there, click on it, *unsupported*.

htrabbit, 2020-10-08:
i fixed mine,  i Installed Prusa Slic3r. problem solved.

konskarm, 2020-10-08:
Something that is marked as "not supported" doesn't mean that it cannot be printed. It just means that Cura cannot determine optimal quality profiles for that because it cannot find preexisting files associated with that material type. So, if you change the material type to "Chocolate", as @Ghostkeeper explains, Cura cannot find quality profiles for Chocolates and it defaults to use very generic default values. 

You can _absolutely_ print your model with these default values (hence why the material can be selected even though it is unsupported), but since they were not optimized for this exact printer-nozzle-material-quality combination, the outcome may not be close to the desireable one. _That's_ what not supported is suggesting. Although I can see how this can be confusing, I don't know how else it could be worded.

On the other hand, if you have tried printing your material with qualities of other materials, e.g. you print eSUN PLA+ with qualities from PLA and it prints fine, then by all means, you can use those. To do that, you can just duplicate eSUN PLA+ and change its material type to PLA. Since Cura actually has quality profiles for PLA, it finds and selects those. Just keep in mind that you may still need to modify many settings to get a good print.

It has been suggested before to change the Material Type field to a drop-down menu, so that the user can select from the existing build-in material profiles to base their material from. Would that be a solution to this confusion?

diamaunt, 2020-10-08:
for eSUN PETG, you provide all the normal profiles, super, dynamic, standard, low, etc.

for eSUN PLA+, all that is throw out and you just say "Not Supported" which makes it *USELESS*.  No hints, no nothing.  "generic PLA" gets various quality settings for who knows what kind of generic PLA.

Why even list eSUN PLA+ as a material then?

If you don't have a profile for PLA+, then simply *DO NOT LIST IT*. just like there's no listing for this whatever brand purple filament I have, I just use generic PLA.

eSUN makes a lot of other filaments, ABS, ABS+ PVA why don't you list all of those and put  "NOT SUPPORTED"??? It seems like PLA+ is in there just so you can look like you have more options... when in reality there is no option.

Listing eSUN PLA+ "Not Supported" makes no sense whatsoever. either simply put in some reasonable (even generic) profiles for it, or just *take it out of the config file*.

It's really quite simple.

htrabbit, 2020-10-09:
wow, they didn't like what i said so they deleted my comment. 
that is commitment to a product

Ghostkeeper, 2020-10-09:
> Why even list eSUN PLA+ as a material then? If you don't have a profile for PLA+, then simply DO NOT LIST IT. just like there's no listing for this whatever brand purple filament I have, I just use generic PLA.

Some people do accept the risk that their print quality will be bad and accept the task to optimise profiles for PLA+.

> eSUN makes a lot of other filaments, ABS, ABS+ PVA why don't you list all of those and put "NOT SUPPORTED"???

We didn't add eSUN's PLA+ and we didn't add their ABS, ABS+ or PVA either. PLA+ was added as community contribution. Why did they add only PLA+ and not the other ones? I'm guessing because they had PLA+ available to them to tune the settings for a different printer and didn't have the others. Or maybe it's because they didn't have time/money to create profiles for all of them.

> either simply put in some reasonable (even generic) profiles for it, or just take it out of the config file.

We can't create reasonable or even generic profiles for your printer and your configuration because we don't have your printer. Sorry! We won't take the material profile out either, since then nobody can print with optimised profiles for eSUN PLA+ any more.

> PS, the whole 'chocolate' thing is ridiculous. there's not a significant difference between 'generic PLA and eSUN PLA+, whereas printing with chocolate is an entirely different ballgame, (and it has been done).

Indeed I was giving "Chocolate" as an example of a material that is clearly of a different type and would clearly require separate profiles. But the difference between PLA and PLA+ is the same according to Cura's profiles. From what we can see, according to [the submission](https://github.com/Ultimaker/fdm_materials/commit/d6a2d11c6e0c04e9ac5a9532196d91255d02c1d5) of these material profiles, eSUN PLA+ is a completely different type of material which requires different profiles than PLA.

It is indeed really quite simple.

diamaunt, 2020-10-09:
I see, you just don't WANT to do anything to improve the usability.

> There are material+nozzle+printer combinations that are unsupported because there are no built-in quality profiles for them. So no, there are no quality profiles that are unsupported. Having a [...]

You provide material+nozzle+printer *quality profiles* that *are supported* for "GENERIC PLA"

your argument falls apart right there, PLA+ is a type of PLA, it falls entirely within the realm of "Generic" because it is *UNSUPPORTED* by your software.

By *YOUR* 'logic' you miight as well have material specs for 'stainless steel wire' "Not Supported" cotton yarn, "not supported" Copper wire +solder, (needs dual extruder, Not Supported" 

> We won't take the material profile out either, since then nobody can print with optimised profiles for eSUN PLA+ any more.

So who can print with optimized eSUN PLA+ profiles?  
Why do you have "optimized profiles" for other things with my printer?

nallath, 2020-10-09:
> wow, they didn't like what i said so they deleted my comment.
> that is commitment to a product

We remove offtopic comments to keep issues from flooding. This isn't a forum, it's an issue tracker. You are indeed right that I have a lot of commitment for Cura. Thank you for noticing! 

> I see, you just don't WANT to do anything to improve the usability.

That is not true. You might not have intended it in this way, but it gives off some vibes that you're trying to start a fight. The fact that we are still responding (and for instance, not locking this issue / deleting it) is a pretty decent sign that we are interested in peoples views about it. 

> You provide material+nozzle+printer _quality profiles_ that _are supported_ for "GENERIC PLA"

Yes, because someone contributed them. 

> Why do you have "optimized profiles" for other things with my printer?

Because people contributed them. 

> So who can print with optimized eSUN PLA+ profiles?

Everyone can print with PLA+ profiles. It will just be less good as the optimized version, but better than not having any such profile. 

You can find some more info aobut it on the wiki: https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings

So having a PLA+ material profile does mean that the temperature settings (and some other settings) are set correctly. It being "not supported" means that it doesn't have a specific quality profile for it (The part that much more fine tuned settings). 

Removing the PLA+ is essentially removing something because the solution isn't as good as it can be. I find that a bit of a weird argument.

htrabbit, 2020-10-09:
I guess my concern is this.....
 i get that Cura can not have all settings for all printers and all filaments, but it could
 **_IF_**
rather than just saying it is unsupported maybe let the user know WHY it is unsupported so we can correct it and make this product even better than it already is. at least i can then have it optimized for my printer and any one else who has my printer.

We have a whole community that use and love this product. lets us then to our advantage and then we can rule the world.

diamaunt, 2020-10-09:
> Everyone can print with PLA+ profiles. It will just be less good as the optimized version, but better than not having any such profile.
>
> You can find some more info aobut it on the wiki: [...]

NO! I can't print with the PLA+ profile, because *ALL IT SAYS* is "not supported" it doesn't list the various suggested settings like even *generic plain PLA* has.    You take away all the profiles and just say "Not supported" why don't you show the generic PLA profiles, or even sone other PLA profiles there and just say "note, not optimized", but no, you REMOVE the profiles. and replace them with "Not Supported" 

Have you not looked at a fresh install of cura? If i click 'generic PLA' I get suggested settings for various quality levels, none of them work great because they all need to be tweaked, but they work well enough.  

When I click 'eSUN PLA+" all of those quality levels *GO AWAY* and there is NOTHING other than "Not Supported".

You are taunting the users "Here's a setting or your purchased filament, on SORRY! NOT SUPPORTED, nya nya nya.

nallath, 2020-10-12:
Yes, you can print with PLA+. All settings that Generic PLA has are settable. What doesn't work is the recommended settings. All the default options are there as normal.

I have looked at a fresh install of Cura. I know how the software that I wrote works, no worries ;)

diamaunt, 2020-10-12:
Yes, I can *print* with PLA+, as long as I don't tell cura that I'm *using* PLA+.

If I tell cura that I have PLA+, then I can't print with it, because there are no profiles, optimized or not.

Ghostkeeper, 2020-10-13:
Again, as far as Cura can see, PLA and PLA+ are different materials. Cura is not going to guess "oh, the names sound similar, it must be the same". It just sees that there is a difference and thus decides that the PLA profiles can't be used. This is on purpose.

If the author of the eSUN PLA+ profile would think that the PLA-specialised quality profiles could be used for that material, he would have chosen to set the material type to "PLA". As a user you can override that, but by default Cura assumes the two materials are as distinct as water and earth.

nallath, 2020-10-13:
> If I tell cura that I have PLA+, then I can't print with it, because there are no profiles, optimized or not.

Yes, yes you can. If there is no profile, you can still _create_ a profile. Just because it's orange (warning) doesn't mean it's impossible. If you just keep the PLA+ profile selected, you can change settings and you can print with it (it just won't be as good as some other materials).

diamaunt, 2020-10-13:
As a user you can override that, but by default Cura assumes the two materials are as distinct as water and earth.

So, and I believe this is the whole point here... 

WHY does CURA show a material that is unsupported?

Nobody seems to be able to address this little point.  If it's unsupported, then don't even show it

let people make up their own profiles if they want to, and don't even show them things that you aren't going to support.

nallath, 2020-10-14:
> let people make up their own profiles if they want to,

I am going to try this one last time. That is exactly why it's there; So people can make them & experiment. We hope that if people find these settings that they share them with us (and the rest of the world)

> No, no I can't because there's NOTHING, I have to start from scratch. just a blank "unsupported" with no settings for various profiles.

There is something. A material can also have settings even if there is no quality for it. You indeed have to start from scratch. But would it be better if we would prevent you altogether from doing this? Or should we break all the setups for people who have it working now? I don't think that that is a good idea.

Anyhow, we are repeating the same statements over and over again, so unless there are any new arguments about it, let's just agree to disagree.

ideal2545, 2021-10-04:
So uh... is this gonna get fixed?

soxhi8, 2022-04-03:
I have the read the full thread, but I still can not understand the point of `Not supported`.
For example:
On my install of Cura, I manually created profiles for filament of the brand Filament-PM. There are *no* presets for Filament-PM. Not for ABS, not for PLA, not for PLA+. For my *manual* profiles, which I manage via Material Settings (for each variation and colour), I get the standard profiles 0.12, 0.16, 0.2 and 0.28mm layer height. This is, what I use the profiles for, foremost and exclusively. Quickly setting my layer height and thus quality level.
Every setting that is specific to the material has to be tuned anyway. Print temperatures, cooling, extrusion and the likes. That is, because Filament-PM does not have profiles in Cura, it is not a supported brand. Still, there are profiles for PLA, which by their nature can not be perfect, because *Filament-PM-profiles* are not shipped with Cura.

When I try the same for PLA+ by Filament-PM, I do *not* get those standard profiles, even though the *whole* brand is not supported. Why is that? Why not just show the specific settings that need to be tuned and still give the different layer heights as templates?

kjkent, 2022-11-13:
This will sound snarky, but, genuinely:

In bug report https://github.com/Ultimaker/Cura/issues/9867, effectively a dupe of this, it's said that this behaviour is 'working as intended' -- but, it begs the question, _working for who?_ I mean, for a new user to Cura, being selected with a raft of built in materials only to be told "none of these are supported"... _who_ actually wants that behaviour? I can confidently posit that new users probably don't, and I'm almost certain materials manufacturers don't. Do experienced users benefit from this UX? If so, fair enough. But if not, it's a genuine question of whether this feature causes more confusion than benefit.

Considering that Cura calculates reasonable defaults for a material such as PLA (and it manages to make this work despite the huge potential combinations of printer hardware), why do the **"supported materials"** not benefit from at least the same support as "generic"?. Forgive me if I'm mistaken, but don't manufacturers materials tend to be _mostly_ variant formulations of these base materials? Would it be possible/sensible to develop profiles that are relative to these?

E.g (I'm making up values here): With an 0.8mm nozzle printing PLA, Cura recommends x, y, z, and calculates other variables based on these, though it's understood that end users may have to tweak settings to match their particular setup. If eSUN comes along and says "here's our PLA+ that requires approx 15C higher nozzle temp, approx -0.5mm initial layer height, etc etc, than the average PLA" (again -- made up, not actual recommendations), could the concept of a relative profile not cater for that by incorporating manufacturer's guidelines with Cura's existing sensible defaults, then compare these to the user's selected printer and its listed capabilities, then flag any warnings based on that?

This is just feedback/an observation, and in my own humble opinion, an outlier in what's otherwise a very user friendly program and a delight to use.

diamaunt, 2022-11-13:
> This will sound snarky, but, genuinely:
> 
> In bug report #9867, effectively a dupe of this, it's said that this behaviour is 'working as intended' but, is it possible to make this behaviour more user friendly, or adapt the current approach to generics, to manufacturer materials?

You couldn't be more right.  the willfully obtuseness of the people 'in charge' of this stance is mind boggling.

If it's *Not Supported" then don't list the material in the default material list.  it's such a *simple* solution but they prefer to confuse, confound and frustrate people.

Ofloo, 2022-12-11:
I think that it is sad, that this still is the way it is. This might sound disrespectful, but we put in more effort describing the issue then the developer has done to fix it. Too me this is just that someone is too lazy to fix it. And rather says it's not supported then to fix their buggy code.

And it was supported before, before no issue now there is. If you change something that breaks something  else that's a bug. If it is or isn't an intended feature is besides the point. To me this says something about the developers.

RRFreemanArchitect, 2023-02-11:
@nallath is just wrong here.  This is simply fixed by defaulting PLA+ to use PLA profiles.

Ofloo, 2023-02-11:
Suggested that very same thing in 2020