Ultimaker/Cura #7277: Update not uninstalling old version correctly when asked to

MarcusWolschon, 2020-03-14:
**Application version**
3.2 - 4.5.1

**Platform**
Windows 10

**Reproduction steps**
1. Install new version
2. tell it to delete all old versions except the previous one.

**Screenshot(s)**
I have a screenshot but I'm still looking for a way to attach it to this ticket.

**Actual results**
Windows 10 reports 3 installed versions but in the
%ProgramFiles% there are still folders for 11 previous versions.
Each supposedly uninstalled one still contains an "uninstall.exe".
That file and folder should have been uninstalled.
Uninstalled programs must not relave anything behind in
%ProgramFiles%

Note that this is %ProgramFiles%. Not %USERPROFILE%. 
So they contain only shared, immutable program code and no user data.

**Expected results**
No remaining folders or files of uninstalled versions.
Leave behind a clean Windows installation with just the versions
that have never been uninstalled.

**Project file**
n/A

**Log file**
No log of installer found.

**Additional information**

MarcusWolschon, 2020-03-14:
How to attach the screenshot it asked to? There is no "attachments" or "upload" button.

smartavionics, 2020-03-14:
[image]

MarcusWolschon, 2020-03-17:
[image]

rburema, 2020-03-17:
Since the msi and the exe are made using slightly different build steps, this might become relevant: Did you install(& uninstall) via exe or msi?

fieldOfView, 2020-03-17:
Is there anything inside those remainder folders?

If there are files in the folder that are not placed there by the installer, the folder will not be removed, because it is not empty after removing the known files. Unless you start Cura "as Administrator", Cura has no write access to those folders, So it should not be able to add any files that are not installed by the installer (and thus removed by the uninstaller).

rburema, 2020-03-18:
Also note that we unfortunately can't fix the older uninstallers -- They may be called from the newest installer, but that calls the uninstallers that where installed alongside the original installations.

We recently fixed a bug in at least the msi installer that should prevent this, but that only goes fro 4.5 onward. (Even then it'll leave those if something else has been placed in that folder, but that's not likely to happen without manual involvement.)

Ghostkeeper, 2020-03-18:
The fix we made was specific to the MSI though, wasn't it? There was no MSI for anything before 4.5. Or did I misremember that this fix was specific to MSI?

rburema, 2020-03-18:
The problem came up for MSI only, and the initial approaches we tried where for MSI as well, but the eventually accepted fix should work for both: https://github.com/Ultimaker/cura-build/pull/233

As you can see though, it's a very specific fix, and will indeed not be the solution if the cause is _not_ the left behind cache files (or one of the other things that was changed for 4.4 or 4.5 w.r.t. the build).

closed 2024-12-07 by github-actions[bot]