Ultimaker/Cura #7278: Export Cura Market matrial for an Ultimaker 2 printer

MarcusWolschon, 2020-03-14:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

When exporting a sliced Ultimaker G-code file to SD card,
there should be a way to get one or all the materials I got from the Ultimaker Marketplace
to the printer somehow.
After all it's the Ultimaker Marketplace in Ultimaker Cura for an Ultimaker Printer using Ultimaker specific g-code that (contrary to regular g-code) doesn't include any material specific temperatures and fan settings anymore because these are to be seleted on the machine.

**Describe alternatives you've considered**
The current documentation talks about using a text editor to write a
materials.txt . The values would have to be genral parameters taken from preferences->materials-><material>->print settings independent of printer or nozzle.
This would have to be done again every time an update is installed from the Ultimaker Marketplace.

**Affected users and/or printers**
Ultimaker 2

**Additional context**

fieldOfView, 2020-03-15:
I think the official stance is going to be that the Marketplace materials are only meant for the Ultimaker 3 and newer.

Having said that, if you change your gcode flavor to "Marlin" (instead of "Ultimaker 2"), you will be rid of the limitations of materials.txt, at the cost of being reminded that the gcode files override settings on your printer.

MarcusWolschon, 2020-03-16:
If I do that (I do when using Octoprint) then in Cura 4.5 all materials vanish (different issue).
That only works in Cura 4.1

Ghostkeeper, 2020-03-18:
> I think the official stance is going to be that the Marketplace materials are only meant for the Ultimaker 3 and newer.

I have a material in the marketplace that supports only Ultimaker 2+.

fieldOfView, 2020-03-18:
> I have a material in the marketplace that supports only Ultimaker 2+

Then this needs fixing.
(dude, I gave you an "out" ;-)