Ultimaker/Cura #7279: Cura 4.5 doesn't saving an filament dia in printer settings

il4masta, 2020-03-14:
**Application version**
Cura 4.5

**Platform**
Win10

**Printer**
Custom FF printer (it's the FLSun Cube actually)

**Reproduction steps**
1. I'm going to the machine settings and trying to change "Compatible material diameter" from 2.85 (which was as default) to 1.75
2. Closing and opening again, dia stay 2.85
3. Even after I've added a new printer and setting the material dia as 1.75 if I'm going to the machine settings again I see the 2/85 dia again 
[image]

fieldOfView, 2020-03-15:
Could you try with a comma instead of a dot? There is (or was?) a similar issue with the nozzle size where using a comma was a workaround.

rburema, 2020-03-17:
What're your 'locale' settings? (More pertinent, what's the decimal separator that's been configured in windows?)

Ghostkeeper, 2020-03-18:
Probably the same issue as https://github.com/Ultimaker/Cura/issues/6546 (but with a different setting).

mituritsyn, 2020-05-15:
4.6.1 Same issue with the filament diameter
Ubuntu 18.04 Russian and English locales.
By deafult it is 2.85 (with dot). Was trying to change to 1.75 or 1,75 - no difference afrer closing it is dropped to 2.85

Thisismydigitalself, 2020-05-21:
Same here.  4.6.1 on Win10/64 keeps defaulting back to 2.85 in printer settings. changed to 1.75 and 1,75 - after restart it's back to 2.85. what's worst is that i saw the issue in printing quality so i checked the gcode and loaded it back to cura and it looks perfect - the gcode / gcode parser does not reflect this issue.

closed 2024-12-07 by github-actions[bot]