Ultimaker/Cura #7280: 2 color first layer problem

leshuguets, 2020-03-14:
I'm using a Geeetech A20M in 2 colour mode using Cura v 4.5.0 on Windows 10

this version of Cura INSERTS the following 3 lines, after it sets temperatures, near the beginning of the Gcode  ...
M163 S0 P0.50 ; set extruder 0 to 50%
M163 S1 P0.50 ;set extruder 1 to 50%
M164 S4 ; save extruder settings 

this results in the first layer ALWAYS printing a 50% mix of colours regardless of the model requirements ... it ruins the first layer of every model
The rest of the model is printed with colours in the correct places.
If I edit the Gcode by removing these 3 lines, all is well. The first layer prints in the correct colours.

Please remove these 3 lines when Cura compiles the Gcode.

Best wishes.

nallath, 2020-03-16:
We can't test this since we don't have a Greetech A20M. As such, it's also hard to say if it will break something else if we were to change it. Maybe @Hrybmo (who originally contributed the script) can say something about this?

leshuguets, 2020-03-16:
you could compile it and send it to me to test !

leshuguets, 2020-03-16:
alternatively please tell me where to edit the script so I can comment out these 3 lines
CheevRS

fieldOfView, 2020-03-16:
Go to the Machine Settings, make the window a bit larger, and you should see the start gcode snippet. Remove the offending lines from there.

leshuguets, 2020-03-16:
thanks ... that should solve my problem.
CheevRS
*_____________________*
*......Mike & Dorrie Norris......*
*.............Les Huguets...............*

*........46310 Frayssinet...........*
      www.leshuguets.co.uk

Ghostkeeper, 2020-03-18:
So what should the start g-code be?

The A10M has the same problem. Given that this printer is advertised online as a mixing printer, I don't dare to remove this from the start g-code blindly.

Hrybmo, 2020-03-18:
I see now... I didn't realize these printers were added to the list, I always used a custom printer. So the complaint is with the default Start G-code, it sets a 50/50 mix. The skirt will use this 50/50 mix and the print may get this mix initially as well, I recommend removing it for a T0 default and using something similar to the following for a default (I purge the filament off the build plate so I left that part out):

M104 S160 ;preheat hotend
M190 S{material_bed_temperature} ;wait for bed temp
G28 ;Home after bed is heated
M109 S{material_print_temperature} ;wait for full temp

If the post-processing colormix script is used then it will set the mix to what is needed for the skirt, purge to a uniform mix, and create the correct first layer on the part. 

I hope that helps,
Thanks

leshuguets, 2020-03-20:
this is the  start script works for my GeeeTech A20M ... it then goes on to
print the first layer correctly
*_______________________________________________*
;GeeeTech A20M Cura 4.5.0  Start Script
G28 ;home
G90 ;absolute positioning
G1 X0 Y0 Z15 E0 F300 ;goto wait position
M140 S{material_bed_temperature};set bed temp
M190 S{material_print_temperature} ;set extruder temp and wait
G1 Z0.8 F200 ; set extruder height
G1 X220 Y0 E80 F1000 ;purge line
;end of start script
_______________________________________________
CheevRS
*_____________________*
*......Mike & Dorrie Norris......*
*.............Les Huguets...............*

*........46310 Frayssinet...........*
      www.leshuguets.co.uk

Ghostkeeper, 2020-03-24:
I've applied that g-code to the next release then: https://github.com/Ultimaker/Cura/commit/a551492ebb8fd6ee3f6ffd65422c3ed5e1540cdd

I've made the following changes to it:
* Removed specification of "Cura 4.5.0" so that we don't have to change it every release.
* Used layer 0 temperatures rather than your temperatures.

However this does remove two things that you might want to keep:
* `M107`, turning fan off, just before heating. Could be used to make it heat faster?
* `M82`, set extruder to absolute mode.

closed 2020-03-24 by Ghostkeeper

MaximeGL, 2020-04-01:
Please be careful M190 set the bed temperature, I think you mean M109

leshuguets, 2020-04-01:
yes  ...its a typo ... M109 is correct
*_____________________*
*......Mike & Dorrie Norris......*
*.............Les Huguets...............*

*........46310 Frayssinet...........*
      www.leshuguets.co.uk

Ghostkeeper, 2020-04-06:
Ah, I see the mistake. I've corrected it for the stable 4.6 version.

azraele100877, 2021-03-17:
@leshuguets I have a similar problem with my a20m, I have sliced it in cura with 2 stl, the problem is that the nozzle stops when it first change the extruder and release a blob on the model in the first layer, then continues to the tower to spurge the second color..I have set yet both extruder temperature at 220 so the machine could recognize that it is the same temperature and doesn't need to wait for arriving at the same. Could you help me figured it out please?

Hrybmo, 2021-03-18:
Most likely there is an M109 "Wait for Hotend Temperature" command, this will be inserted regardless if the temps are the same (last time I checked).
You can comment this out using a text editor or use "TwoToOneHotend.py" that I created on Thingiverse to help automate temperature issues- https://www.thingiverse.com/thing:3442615/files
Ideally this should be a Cura "Multiple extruder" option and not a post-process script.

azraele100877, 2021-03-18:
Thanks for the links I will use it when mixing, but my problem is with 2 colors print and a 2 color tower next to it, my stl is thin, made from 2 stl, a vampire skull  in grey and the other stl for the background in black. Did it in thinkercad, exported both, opened in cura I aligned them and I gave each stl an extruder..well it stops the first time that from the background should go to the tower leaving a blob on the print!! Am I doing the procedure wrong? Should I search for this command you told me and eliminate it from the code?

Ghostkeeper, 2021-04-07:
It sounds like that printer profile is not properly tuned then.

I would advise to configure the `machine_extruder_end_pos_x` and `machine_extruder_end_pos_y` settings, with a small relative position such that it moves away from the print, or with an absolute position such that it moves to a fixed location on the build plate to switch there (if the switch takes time). These are machine settings though, so only available in the .def.json files or through installing the Printer Settings plug-in from the Marketplace.