Ultimaker/Cura #7281: [4.5.0] Importing ANY Material (including Marketplace) WIPES Profiles

PuterPro, 2020-03-14:
Please note up front that is related to / replaces #7267 and #7250 , but is a more accurate report of the problem.

I'm a PC tech and have spent a bunch of time troubleshooting this to help you narrow down the EXACT problem, please forgive if I've erred in creating a new bug report to replace those two as I found it's REALLY one issue. I've uninstalled and reinstalled 4.5 EIGHT times to chase this, so be gentle on me. :-) 
#7250 has the most info on this, to get you up to speed.

**Application version**
4.5.0

**Platform**
Win10 latest updates. nVidia GTX 2070 Super, same.

**Printer**
Originally Customized CR-10S, but I repeated tests with the built in CR-10S, default settings.
For fun I also tried Creality Ender 5, same results.

**Reproduction steps**
1. Import a new Material [Tried MatterHackers NylonX] from the Marketplace, **_says successful_**. 
2. Tried selecting from drop down. NOT LISTED. 
3. Examined Manage Materials, NylonX not listed despite "Successful" report.
4. Imported a self created Material from an Exported 4.3 **working** material. Successful import.
5. Selected from drop down. Loads but WIPES THE STANDARD PROFILES and puts "Not Supported" in it's place. Also fails to update the 4 Nozzle Temps, but correctly imports rest of settings like bed temp.

**Screenshot(s)**
[image]
[image]
[image]

**Actual results**
If you load ANY custom material all Standard Cura Profiles goes "Baby Says Bye-Bye".
Any ADDED Profiles will remain (not shown), but will only show the CURRENT profile in Drop down, others appear in Manage Profile (but **no** Standard Cura Profiles, as shown). ie: Five loaded Profiles will show in Manage Profiles, but only ONE shows in drop down. Odd.

**Expected results**
Materials should load, not wipe Profiles.

**Project file**
Not related to models.

**Log file**
[Materials Wipe Profiles 4.5 Config.zip](https://github.com/Ultimaker/Cura/files/4333465/Materials.Wipe.Profiles.4.5.Config.zip)

**Additional information**
I'll be happy to help if you need me to try a new thing, but as I said, I've spent most of an afternoon and eight reloads to chase this ... Kinda surprised to see this big a bug get out. Software - can't kick it when it's naughty.

MarsupialVomit, 2020-03-15:
I second this behavior. I just came here to see if this has been addressed yet.

nallath, 2020-03-16:
It doesn't wipe it. It just tells you that it doesn't have specific settings for that printer. The marketplace essentially only has profiles for Ultimaker printers. So although you can download the models (and use them to some extent on other printers), you won't get the full benefit as you would when using them combined with Ultimakers.

So a bit more in depth; Cura has a few types of profiles it can have. Quality profiles ("What does normal quality for a material look like") and a material profile ("What does PC-MAX look like at all"). In this case, there is a material profile for your machine, but not a quality profile.

PuterPro, 2020-03-18:
@nallath - So this isn't a bug as much as incomplete machine profile, eh? I didn't get that relationship. 

Having a list of compatible printers would indeed be a blessing for the millions of Cura users who can't afford to use Ultimaker's awesome machines.

I've updated the other posts, and for those of you who were following this, @Ghostkeeper 's answer in #7250 also explains some more. Some additional related reading in post #7267 .

Thx SO much for your time and answer @nallath your work is GREATLY appreciated. This stuff gets complex, and the team supporting here do a rockin' job helping us all. :-)

closed 2020-03-18 by PuterPro