Ultimaker/Cura #7282: [4.5.0] Failure Midprinting Almost Constantly

TheGangster1023, 2020-03-15:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-03-16:
So if I understand correctly, the printer freezes sometimes when it's connected via an USB thether (and it doesn't freeze when using an SD card)?

TheGangster1023, 2020-03-16:
I haven’t been able to try using an SD card yet because the adapter I ordered to I could write to it hasn’t come in yet.

rburema, 2020-03-17:
Is it always happening on the same layer?

Ghostkeeper, 2020-03-18:
Maybe this happens when the computer goes into sleep mode?

TheGangster1023, 2020-03-18:
Everytime I’ve had it happen I have been actively doing something on my PC, so there would be no way that could be it and I have it were it never sleeps in my Windows settings.

TheGangster1023, 2020-03-18:
It doesn’t seem to happen until late into the print but I don’t believe its on the same layer.

Ghostkeeper, 2020-03-19:
Some more possible causes are written down in the Settings Guide: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/unfinished_print.md#printer-failures

Cheaper printers seem to sometimes get their stepper drivers to fail because they become too hot.

As it stands, we're not really able to help you because we can't pinpoint where Cura is doing it wrong. When I'm making a print it finishes every time (unless I abort).

TheGangster1023, 2020-03-19:
Where would the stepper driver be located? I would be fine with modding my printer to have another fan.

Ghostkeeper, 2020-03-24:
On my printer it's towards the edge of the electronics board. Built into the board design.

Here's an article on the RepRap Wiki with some example pictures: https://reprap.org/wiki/DRV8825

closed 2020-06-19 by no-response[bot]