Ultimaker/Cura #7283: Printing on Ultimaker 2+ with Cura 4.5.0 not working

gcraenen, 2020-03-15:
**Application version**
4.5.0

**Platform**
OS X latest version 

**Printer**
Ultimaker 2+

**Reproduction steps**
1. Reset UM to factory settings, installed fresh version of Cura 4.5.0, connected to USB, upgraded UM 2+ to latest firmware from Cura.
2. Started a testfile UM2_Tetra_wire

**Actual results**
extruder starts to move if it is heated and the printed is at the right hight. But extruder and printbed are not heated up and nothing is printed.

**Expected results**
Printed tetra-wire test

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**

fieldOfView, 2020-03-15:
USB printing has never been officially supported (or recommended) with the UM2 family. If you want to print via USB on an Ultimaker 2+, you need to change the gcode flavor to "Marlin" (from "Ultimaker 2") in the Machine Settings on the Printers pane of the Preferences.

gcraenen, 2020-03-15:
> USB printing has never been officially supported (or recommended) with the UM2 family. If you want to print via USB on an Ultimaker 2+, you need to change the gcode flavor to "Marlin" (from "Ultimaker 2") in the Machine Settings on the Printers pane of the Preferences.

Changing that settings does not change anything, unfortunately.

Ghostkeeper, 2020-03-18:
The actual results are indeed our expected results. The USB port is only there to allow you to do firmware updates. I've had success with USB printing by changing the g-code flavour to RepRap (rather than Marlin). Alternatively you can add some pre-heat and fan speed commands to the start g-code.

The problem is that the "Ultimaker 2" g-code flavour specifies that there should be no temperature commands (nozzle or bed), no fan commands and no soft retractions in the g-code. Instead, the printer gets the correct temperatures and fan speed from its own built-in material profiles and also the retraction length and speed that it uses for the firmware retract commands (G10/G11) in the g-code. When printing over USB, Cura has still generated the same g-code. When this g-code is blindly executed, the printer doesn't know that it should also pre-heat to the printing/bed temperatures of its current material.

closed 2020-03-18 by Ghostkeeper