Ultimaker/Cura #7284: FEATURE REQUEST: Multiply object as a grid

tlhintoq, 2020-03-15:
Setting up lots of objects on a bed is long drawn-out workflow, if you don't want some weird random placement.

Some sort of "Matrix Multiply" would be great; where you feed X and Y spacing.  The attached screen shot is 90 items with x & y of 30 each.  That bed of objects took just over 30 minutes to layout.

- Bring in 1 object
- Multiply it one time
- Manually set the X and Y
- Repeat 8 more times to make one row

- Group all of those
- Manually multiply it 1 time
- Set the X and Y
- Repeat it 4 more times to make a half bed 
- Group all
- Manually multiply 1 time 
- Set the X and Y
- Finally have a full bed

OBSERVATION: With every paste of a single row of objects the waiting time gets longer and longer, until you can grab the row and set the position.  Which seems a bit odd for an 8 core i7 with 32gig of RAM and seriously fast dual SSD drives, running no other apps but Cura while doing this work.
Automatically slice is NOT ticked on: So it shouldn't be doing exponentially more work with each iteration of adding objects.

In the end, repeating 1 group of 4 rows is a lot faster than repeating 1 row, 4 times.  
And with 90 items on the bed, every action is very very slow.
Select all...
Wait 20 seconds
Click in the X location field
{not responding for 20 seconds}

So being able to define a matrix instead of just "Multiply 70 times" without knowing in advance just how many could fit - and resulting it strange placements - would be a great boost to shop efficiency.

[image]

Do something while its busy
[image]

[image]

monkeymademe, 2020-03-16:
If you used silhouette studio they do this grid multiply! its very handy and missing from cura

nallath, 2020-03-16:
Re-working the arrange algorithm is pretty high up on our backlog on our internal tracker (CURA-7259)

tlhintoq, 2020-03-16:
> @nallath 
> Re-working the arrange algorithm is pretty high up on our backlog on our internal tracker (CURA-7259)

High up... Meaning towards the top of the list... high-ish in priority... So might see some love sooner rather than later?

tlhintoq, 2020-03-16:
> 
> @monkeymademe 
> If you used silhouette studio they do this grid multiply! its very handy and missing from cura

Hadn't heard of of silhouette studio.  I'll take a look.  Thanks for the tip

nallath, 2020-03-16:
> > @nallath
> > Re-working the arrange algorithm is pretty high up on our backlog on our internal tracker (CURA-7259)
> 
> High up... Meaning towards the top of the list... high-ish in priority... So might see some love sooner rather than later?

Assuming nothing changes, I'd say within the next few sprints.

tlhintoq, 2020-03-16:
@nallath 
**Outstanding!**
I'll be on that beta as soon as I see it.  If you want a tester faster than that - hit me up.

tlhintoq, 2020-03-17:
Side note:
The same PC that created this file yesterday - slowly with that many objects - cannot open it today.  Cura locks up.  Have to kill it in Task Manager.

So now I have to remake the file rather than make one little change.  Clearly I'll do it with less objects.

Ghostkeeper, 2020-03-18:
We've made a few improvements to how Cura handles many objects on the build plate in v4.5, but it's not quite there yet.

tlhintoq, 2021-09-15:
>  nallath commented on Mar 16, 2020
> Re-working the arrange algorithm is pretty high up on our backlog on our internal tracker (CURA-7259)

Just wondering... Its a year and half later for this "pretty high on our list" ticket.
Time to change icons, but no time for features?

nallath, 2021-09-15:
Well, we did re-work the arrange algorithm. Previously we had a completely self built algorithm that did a far worse job than the one that we're using right now.

Yes, we did feel that updating the icons a bit was more important than making sure objects were placed in a grid. 

Feel free to have a look at our current implementation and check if you can improve them. We're always open to pull requests (and I'm sure that Prusa Slicer is as well; The changes could be contributed back to them as well)

alexinnnn, 2022-01-01:
https://github.com/alexinnnn/CuraMultiplyByGrid
I maked plugin to multiply selected model by a grid. It doing it automatically with specified distance and maximum pieces.

Drumdevil, 2022-01-22:
> Well, we did re-work the arrange algorithm. Previously we had a completely self built algorithm that did a far worse job than the one that we're using right now.
>
> Yes, we did feel that updating the [...]

It feels a bit counterintuitive still having to rotate them back and place them properly, wasting a lot of time. Imagine having to make a grid of 40 pieces, having to manually calculate the X/Y position of each and rotate them back after multiplying them. 

> https://github.com/alexinnnn/CuraMultiplyByGrid I created plugin to multiply selected model by a grid. It doing it automatically with specified distance and maximum pieces.

You are a life saver, thank you!

MariMakes, 2024-01-11:
Quick update from our side 👋
We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue about multiplying objects in a grid. 

This functionality is introduced in Cura 5.5 last October. If you download the latest version of Cura you can try it for yourself. 
[image]

I'll close this issue since there is a solution in place. 
Thanks for your report. Together we can make Cura better 😊

closed 2024-01-11 by MariMakes