Ultimaker/Cura #7286: reload / replace objects when opening a 3mf project file

rooiejoris, 2020-03-16:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I printed something then I made a iteration of the object, I reopen the 3mf file and then I cannot replace the specific object with the new version. I save project settings in project files, when the file is open, you can 'reload' the [modified] stl when you noticed something that would slice nice and adjust the stl. But when you open the 3mf, it is impossible to replace the file since the 'origin' is not the same anymore as the origin in the newly changed exported stl.
Up to now I delete the object and load the new model, but then it looses the orientation and location made in Cura. Also when using per object settings, you have to set those setting again [this is maybe the most annoying!]. For printing a single object the problem is not that big, but when using the advanced functions of Cura, like importing objects to print as support the problem is bigger and bigger.

**Describe the solution you'd like**
Would be so nice and productive to be able to replace the current STL/object with keeping/respecting the original origin/scale.

**Describe alternatives you've considered**
I can rotate and orient the objects in CAD, but this is not nice since the orientation in CAD/Assemblies is not alwasy [or always not?!] the same orientation best for printing FDM.

**Affected users and/or printers**
This will improve the workflow so much for users who design for FDM. More thecommercial/ industrial users who has to make iteration of the same model over time.

**Additional context**
There is a recent change in the plugin Mesh Tools which made it possible to change the object, but its quite hidden and i think it should be located in the rmb menu of the object.

nallath, 2020-03-16:
We recently got a pull request that makes some changes in this behavior, but not all of them. Could you have a look and see those changes would solve your issue? See https://github.com/Ultimaker/Cura/pull/7230

rooiejoris, 2020-03-17:
I don't understand what exactly is fixed at #7230 
What I mean is that reload [or replace] doesn't work when opening a .3mf projectfile.

nallath, 2020-03-18:
What that fixes is that the meshes inside the 3mf do get reloaded, but only to a limited extent. So if you rotate the mesh in hte other application, that will not be updated. If you actually change the geometry, it will be updated.

fieldOfView, 2020-03-18:
That PR is more aimed at loading 3MF files as models than with loading 3MF files as projects. Perhaps there should be more of a difference between the two.

fieldOfView, 2020-03-20:
The next release of [MeshTools](https://github.com/fieldofview/cura-meshtools) will let you right-click a model and replace it with a different file.

[This PR](https://github.com/Ultimaker/Cura/pull/7310) will at least retain the model _names_ when saving and loading projects. A followup to that PR could do something similar to (optionally) save filenames in 3mf project files, which would allow reloading models within that project.

kc2zgu, 2021-08-12:
I've wanted to do this exact same thing - arrange several objects in a project, possibly with some custom settings or just multiple pieces that make up a larger assembly so it takes more effort to replicate than just loading one object, save it, later modify one of the models (and hence recreate the STL file), and be able to open the project and replace a mesh with the new file. Ideally it might be better to have a separate project file format since the idea of 3MF AFAICT is to embed all the mesh data rather than referring to existing files, so it really just takes up extra space for this use case.

(Can't try MeshTools since for some reason Marketplace is completely non functional on my system and I don't know of anoher way to manually install plugins)

nallath, 2021-08-12:
@kc2zgu Using my amazing marketplace admin powers, i've downloaded the .curapackage for you.

Unpack this zip file (Github doesn't like the .curapackage extension) and drag that into cura. Reboot cura and the plugins should be there. 
[MeshTools.zip](https://github.com/Ultimaker/Cura/files/6975956/MeshTools.zip)

Some developers (aka; Me) also upload the cura packages in the "releases" tab of their github repos

B-Man1, 2024-02-10:
I second this.  It would be great if re-load file actually fully reloaded the .STL files within a project file.  Meshtools-replace model does not include the rotations so if you have things like support blockers in your model, they will not be in the same place if you revise and reload the .STL file