Ultimaker/Cura #7287: USB link makes my printer Beep loop and not usable

MrCanan, 2020-03-16:
Hi,

I update my Ultimaker Cura to lastest version (because it always ask me) and I have now a problem (I already got when I installed the previous version).

The problem is the Printer works alone (move and print), but when I plug USB inside and launch Cura, my printer starts to beep in the loop and make only a strange move.
The solution for the previous version was to edit a Cura program file to comment on some parts, but I forgot which one.

I can't find the post on the internet that solved it before (if it is the same). ^^
Can someone help me?

Thank you in advance,
Best regards.

NB: I already try with another one, uninstall and install again, and especially it works before making the update.

fieldOfView, 2020-03-16:
Go to the "Installed" tab of the Marketplace, and scroll down to "USB Printing". Uncheck that checkbox and restart Cura.

MrCanan, 2020-03-16:
Thank you for your help,
For sure, if I disable the USB plugin, I will not have the problem. ^^
But it not solves it because it makes Cura unusable via USB (as I print with it, so I need it).

I remember something like editing a Python in "C:\Program Files\Ultimaker Cura 4.5\plugins\USBPrinting" folder.
Can someone help me?

NB: My printer is the Ender 3 Pro.

closed 2020-03-16 by MrCanan

fieldOfView, 2020-03-16:
I hope my answer did not make you go away.

If you solved your problem by yourself, it is more customary (and helpful for others) to share how you fixed it, than to remove your post.