Ultimaker/Cura #7289: 4.4.0 -

Elocai, 2020-03-17:
**Application version**
(The version of the application this issue occurs with.)
4.4

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
PC Win10

**Printer**
(Which printer was selected in Cura?)
Not related, CR10

**Reproduction steps**

1. Enabled Tree Support
2. Set "Support Overhang Value" to a very low value like 0
3. Enabled "Enable Support Roof"
4. Set supports to print "Everywhere"

**Screenshot(s)**
[image]

**Actual results**
(What happens after the above steps have been followed.)

Supports Roofs are generatated EVERYWHERE even where the Tree Support can´t reach the object

**Expected results**

Support Roof are generated ONLY where Tree Support is actually present

**Reason**
Support Roofs are implied to be part of the support structure and their task is to support an object,
adding support roofs everywhere is not very helpful as they are not supported by tree support that way

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

Every Organic Structue (standing figures) would work

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
not related

**Additional information**
(Extra information relevant to the issue.)
everything present in saying

rburema, 2020-03-17:
I'm pretty sure this is a duplicate of #7050 (and #7131). Please continue discussion there (the first one), or reopen if you disagree.

closed 2020-03-17 by rburema

Ghostkeeper, 2020-03-19:
(Users can't reopen an issue if a member like rburema closed it. They can still react though.)

rburema, 2020-03-19:
(Some day I'll actually remember that when I write these replies.)

nallath, 2020-07-13:
This has been fixed for 4.7