Ultimaker/Cura #7290: It is recommended to add the function of real-time preview / modification of G code.

gailulun, 2020-03-17:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

fieldOfView, 2020-03-17:
This would be hard to implement.

What Cura normally displays in layer view is not actually based on the gcode, but is specifically created data for layer view. Changing the gcode does not change that data.

Have you ever saved a gcode file, and loaded that gcode file back in? That take *much* longer than displaying the layers directly after slicing. This is because parsing gcode line by line and converting that into layer geometry is "hard work" for Cura.

gailulun, 2020-03-17:
I think we can just make an interface. The code display is an option. After clicking the "Generate Code" option, the display box pops up. After modifying the code, click the "Preview" button to generate a preview. At least we don't need to use Notepad to change the code and save it, then use cura to open the modified file step by step.
//我想我们可以只是做一个接口。代码显示是一个可选项，点击“生成代码”选项后弹出那个显示框，修改完代码后再点击“预览”按钮，再生成预览。至少我们不需要用记事本改完代码并保存后再用cura一步步的地去打开那个修改好的文件了。

gailulun, 2020-03-18:
If possible, I think it can be displayed while generating G code, similar to video loading.

gailulun, 2020-03-18:
[image]

rburema, 2020-03-18:
I'm sorry, but I've discussed this with other developers, and, due to the amount of work and our priorities, we've chosen to defer this for now.

Pull-requests are of course welcome.

gailulun, 2020-03-18:
I understand you very well, and being able to put it on the deferred list is also very good, at least one day this function may be implemented instead of forgotten. ；）

rburema, 2020-03-18:
Thanks for understanding!