Ultimaker/Cura #7291: Better support edition

erickwill, 2020-03-17:
What I'm suggesting is a improvement in a current plugin Custom Supports and Support Blocker, which I do believe that could be integrated into the Cura Core. I'd like to suggest the ability of edit, add and remove manually, the supports beyond the edit the automatic support creation. Idea maker already uses this neat feature besides simplify3d. This cool and useful feature is the only thing I missing while using Cura. Thanks for doing the best slicer ever.

Ghostkeeper, 2020-03-18:
Being able to customise the support placement is a long-time wish of the Cura developers as well. We made some strides for that once, offering a paint brush tool for the red areas on the model that get painted with support. But we never finished the implementation. What was missing is an eraser tool so you can also undo your operations as well as remove support that was previously generated.

This tool also requires some thought in how it interoperates with the current "Overhang Angle" setting. After you customise the support, what should happen if you change the overhang angle?

erickwill, 2020-03-18:
> Being able to customise the support placement is a long-time wish of the Cura developers as well. We made some strides for that once, offering a paint brush tool for the red areas on the model that get painted with support. But we never finished the implementation. What was missing is an eraser tool so you can also undo your operations as well as remove support that was previously generated.
> 
> This tool also requires some thought in how it interoperates with the current "Overhang Angle" setting. After you customise the support, what should happen if you change the overhang angle?
> 
> The customising support plug-in is still on a branch here: https://github.com/Ultimaker/Cura/tree/feature_custom_support/plugins/CustomSupport

Thanks for your prompt reply and explanation.
Yes, now I understand the challenges involved into achieve the correct implementation for this feature,
I just wonder about the "Overhang Angle" settings, if a viable way to work with along with the Custom Support would be to lock the variable overhang angle in a safe value to get custom support working if and when custom support is in use. I wonder if you disable overhang angle while custom support is in use would be, also, a viable way..
Once more, thanks for your hard work and time.

Triangulix, 2020-07-09:
Hi
I found this request because I noticed an unexpected behavior in the custom supports plugin:
My expectation was: I set Cura to create support where it touches the buildplate and then manually add some support in the model.
Unfortunately it seems that Cura in this situation only creates the walls of the manually added support but not the surfaces where the support would touch the supported parts.
This makes it a problem to print an air duct that needs support inside of some long slots but of course cannot be filled up with support inside.

Ghostkeeper, 2020-07-10:
Which custom support plug-in? The one that I linked here in the above reply, or the one in the Marketplace?

The plug-in I linked here was an experiment several years ago. It's not expected to work correctly in the current Cura version. The plug-in in the Marketplace is maintained by Lokster and issues should be reported there: https://github.com/lokster/cura-custom-supports/issues

Triangulix, 2020-07-10:
> Which custom support plug-in? The one that I linked here in the above reply, or the one in the Marketplace?
I meant the PlugIn in the marketplace.

Ghostkeeper, 2020-07-13:
Then please report your bug here: https://github.com/lokster/cura-custom-supports/issues

userosos, 2020-08-06:
No! It is very bad! If i use the plugin i have pain. See the pics. it is crazy. I need scale the box i has notime for it. I need ighlight those surfaces on the model that I want.
[image]
[image]

I mean that i dont want use box! I mean that i want hold down the mouse button and paint the model. The surfaces that I painted will have supports. See the pics.
[image]
[image]

I hope you understand me now.

Ghostkeeper, 2020-08-07:
I understand precisely what you mean. And indeed that is exactly what my own [custom support](https://github.com/Ultimaker/Cura/tree/feature_custom_support/plugins/CustomSupport) plug-in did. I still have a demo video of it here: http://dulek.net/work/custom_support_short.mp4

This plug-in is not to be confused with the "Custom Support" plug-in by Lokster which only places a cube in the scene similar to the support blocker tool. As you can see in that video, my plug-in was made for a fairly old version of Cura. Really I'd like nothing more than to finish that plug-in, believe me. But I can't.

userosos, 2020-08-07:
> I understand precisely what you mean. And indeed that is exactly what my own [custom support](https://github.com/Ultimaker/Cura/tree/feature_custom_support/plugins/CustomSupport) plug-in did. I [...]

Very GOOD! Thx! Can i use it rightnow?

Ghostkeeper, 2020-08-07:
I expect that plug-in will not work in modern versions of Cura. And it was limited to begin with, with a number of hacks. It is not fit for release.

You can try copying the source code in, but no promises.

userosos, 2020-08-07:
> 
> 
> I expect that plug-in will not work in modern versions of Cura. And it was limited to begin with, with a number of hacks. It is not fit for release.
> 
> You can try copying the source code in, but no promises.

It is not work. Sorry. Do you have some plan about it?

erickwill, 2020-08-07:
WOW.. such a plugin would be AWESOME! :)

Ghostkeeper, 2020-08-10:
> It is not work. Sorry. Do you have some plan about it?

Indeed, like I said, I expected it to not work. I have no plans to do anything with it unless I can do that during work hours.

nallath, 2020-11-19:
Duplicate of #5955

closed 2020-11-19 by nallath