Ultimaker/Cura #7292: On some models, the generated code causes the nozzle to crash into the print surface

mjskier, 2020-03-17:
**Application version**
4.5.0

**Platform**
Linux Mint 19.3

**Printer**
Creality CR-10s Pro

**Reproduction steps**
1. Sliced Body_cap1.stl from this link: https://www.thingiverse.com/thing:4222048/files/Body_cap1.stl

2. While printing the skirt, the nozzle crashed into the bed  hard enough to leave gouges. The skirt started fine, (3/4 of it was ok before the crash)
The 2 gouges closest to the front aer this time. The other 2 from the other time the problem happened.

[image]

My bed is essentially flat, so there is no reason why the nozzle would crash at a specific location instead of at the start, or at the first side.

[image]

The same thing happened when I tried to print 4 copies of a small model that printed fine on its own.

**Expected results**

At a minimum, even if there was an issue with the model, I would expect the skirt to print correctly, not ended trying to print below the surface.

**Project file**
https://www.thingiverse.com/thing:4222048/files/Body_cap1.stl

**Additional information**
I'm using the default cr-10s-pro printer profile, and the default Hatchbox PLA profiles that came with Cura 4.5.0.

I am attaching the generated gcode (renamed to .txt to allow the upload)

[CCR10SPRO_Body_cap1.txt](https://github.com/Ultimaker/Cura/files/4345357/CCR10SPRO_Body_cap1.txt)

smartavionics, 2020-03-17:
I looked at your gcode, the entire first layer is printed at the correct height (0.2mm).

So what I think is more likely to be the problem is either the z-axis is not operating correctly or this line in your gcode is having a bad effect:

M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height

What happens if you don't enable ABL?

mjskier, 2020-03-17:
Thanks for the quick reply. I'll give it a try. 

Maybe related to another problem I ran into on a different model: Instead of crashing into the bed, the gantry goes all the way to the top (and doesn't stop, I have to turn the printer off)
There too I didn't see anything in the gcode that would make it do so.

mjskier, 2020-03-17:
No crashing when I comment out the M420. So it looks like the problem is on my end.
I did a "M420 T0 V1" which I think is how to get the value of the mesh from the printer and I get this, which doesn't match what is on the screen. That +160112.43750 doesn't make any sense.

Recv: Bilinear Leveling Grid:
Recv:         0        1        2        3        4
Recv:  0 +0.00000 +0.00000 +0.00000 +0.00000 +0.00000
Recv:  1 +0.00000 +0.00000 +0.00000 +0.00000 +0.00000
Recv:  2 +0.00000 +0.00000 +0.00000 +0.00000 -0.00000
Recv:  3 -0.00000 -0.00000 +160112.43750 +0.00000 -0.00000
Recv:  4 +0.00000 +0.00000 +0.00000 +0.00000 +0.00000

Anyway, thanks for the suggestion. Closing this ticket since isn't Cura's problem.

closed 2020-03-17 by mjskier