Ultimaker/Cura #7293: cura 4.5  does not set the correct material temperature...

manvalan, 2020-03-17:
cura 4.5

W10
Geeetech A10

Cura 4.5 do not set the correct material temperature but reset to 215 (extrusor) 55 (plate)

Vandrasc, 2020-03-17:
Hi @manvalan , could you please fill in the bug template, this way we will receive enough information in order to help you. Thank you!

Ghostkeeper, 2020-03-19:
I'm getting 200:
[image]
That's being defined by Generic PLA, so it seems to be correct.

Alvgon, 2020-03-19:
May be he has still active the post processing scripts installed with v4.5 Winx64, changing extruder temp every 25 layers ...

fieldOfView, 2020-03-20:
Or, more likely, he has a quality_changes profile that includes a temperature which overrides whatever is set in the material. We won't know unless @manvalan posts a project file.

fjgo, 2020-03-22:
Hi.
I faced the same problem and the error is in the profile, in the **Start G-Code** the temperature settings are included.

mahtDFR, 2020-03-23:
I can confirm it's 215 by default in the profile:

[image]

As a workaround I'd suggest altering the start Gcode.

We don't maintain third party machine profiles and are reluctant to do so, so I we won't make any fixes to the definition from here.

fieldOfView, 2020-03-23:
That's a bad workaround. Because the {material_bed_temperature_layer_0} and {material_print_temperature_layer_0} replacement patterns are not found in the start gcode, Cura will prepend its own heat up sequence. So you end up having two heatup sequences in the start gcode.

manvalan, 2020-03-23:
In Cura 4.5 for OS X the setting of material bed temperature and the extruder material temperature are correct… it seems to be a bug in the windows version

fieldOfView, 2020-03-23:
> it seems to be a bug in the windows version

I doubt that. It is much more likely an issue with a setting or a stored profile. We won't know unless you save a project (File -> Save as) and upload it here (you may have to zip it first).

Ghostkeeper, 2020-03-24:
Indeed, Cura's printing/bed settings don't have any effect since it's being overwritten by the start g-code.

Would you like to make an adjustment here, @amitlobo ? You were the author of these profiles. We can't test any changes that we make here, so I'd like a user of Geeetech printers to make these changes.

Ghostkeeper, 2020-03-24:
This is probably fixed for the M10 and M20 here: https://github.com/Ultimaker/Cura/issues/7280

Jaki1091, 2020-04-14:
I have the same problem with my Geeetech A10 (Marlin 1.1.8). 
This is my start g-code in Cura 4.5:

G28
G1 Z15 F300
M107
G90
M82
M104 S215
M140 S55
G92 E0
M107
G0 X10 Y20 F6000
G1 Z0.8
G1 F300 X180 E40
G1 F1200 Z2
G92 E0
G28

No matter what material/temperature I set, the printer will only print at 215°C. Any clues why?

mahtDFR, 2020-04-14:
> M104 S215

This is a gcode command to heat up the nozzle to 215 C. Try changing it to a different value.

[This](https://reprap.org/wiki/G-code#M104:_Set_Extruder_Temperature) is a useful resource.

Jaki1091, 2020-04-14:
@mahtDFR yes, I am currently changing that value in order to print different materials, but I guess it's not the correct way. Whatever material I select (ABS, TPU, PC, PLA) or whatever temperature I set in the "material" tab, the printer will always print with the value written in the start g code (i.e. M104 S215)

mahtDFR, 2020-04-15:
What if you substitute this:

>M104 S{material_print_temperature}

If I have got this correct, then it will take the value from your material print temperature and substitute it here to the nozzle heat up sequence in your start gcode.

Jaki1091, 2020-04-15:
@mahtDFR I tried what you said but the printer didn't extrude anything (the extruder gear did't rotate).
This the gcode Cura created after the edit in the start gcode (I set the temperature at 220°C in the material tab)

M140 S60
M105
M190 S60
M82 ;absolute extrusion mode
G28
G1 Z15 F300
M107
G90
M82
M104 S220	
M140 S60
G92 E0
M107
G0 X10 Y10 F6000
G1 Z0.8
G1 F300 X180 E40
G1 F1200 Z2
G92 E0
G28
G92 E0
G92 E0
G1 F2400 E-4

mahtDFR, 2020-04-16:
Most of these E moves are 0 except for these two:

> G1 F300 X180 E40
> G1 F2400 E-4

So there should be something happening there at least. Unfortunately I dont have a GeeTech machine to test. If you remove the substitution, does it stop this behaviour?

Jaki1091, 2020-04-16:
@mahtDFR thnk you for the support, just solved the problem by removing from the start gcode 
M104 S215
M140 S60
I thought that by removing these two comands the printer would have extruded at room temperature, instead it reads the temperature set in the material tab.
Cheers

mahtDFR, 2020-04-16:
Ah, good to hear you have a solution!

Hotled, 2020-04-30:
This is not a fix.
It is an issue if your using a Geeetech a10. 
Copy all your settings and use a profile from an Ender3.
Something has bugged the Geeetech profiles.

mahtDFR, 2020-05-01:
> Something has bugged the Geeetech profiles.

Maybe you can share a project file with me so I can reproduce your issue?

mahtDFR, 2020-05-01:
@Hotled We would ask that the contributor makes a fix for this, we don't maintain the profiles and can't test them either. Is it just the start gcode or is there more?

closed 2020-06-19 by no-response[bot]