Ultimaker/Cura #7294: cura4.5  is in conflict with marlin2.0

hn0550, 2020-03-18:
**Application version**
cura 4.5 

**Platform**
windows10 + I3 8100

**Printer**
DIY corexy printer  firmware：marlin2.03

**Additional information**
cura4.x slice files ，Only the first layer can be printed，marin 2.0 to crash when changing layers。but  firmware is marin 1.1x ，print normal。

konskarm, 2020-03-19:
Could you please write the detailed reproduction steps that causes the crash?

smartavionics, 2020-03-19:
Hi, I am using Marlin 2.0.4 with current Cura and it works just fine.

Please save your project file (File -> Save) and then either zip up or rename the .3mf file and attach to this issue. Thanks.

hn0550, 2020-03-19:
> 您能写出导致崩溃的详细复制步骤吗？

thank you，I'll reinstall it ，Have normal。

hn0550, 2020-03-19:
> 嗨，我在当前的Cura中使用了Marlin 2.0.4，它工作得很好。
> 
> 请保存您的项目文件（文件->保存），然后压缩或重命名.3mf文件并附加到此问题。谢谢。

thank you，I'll reinstall it ，Have normal。

Ghostkeeper, 2020-03-19:
Okay, then we'll assume that this issue was a glitch in the installation I guess. We won't be able to reproduce the error any more.

closed 2020-03-19 by Ghostkeeper