Ultimaker/Cura #7295: [4.5.0] Firmware Retract Missing?

CCS86, 2020-03-18:
Am I just missing something, or is the firmware retraction option gone?

I have even tried creating a new, custom FFF machine and still do not see it.

DivingDuck, 2020-03-18:
You are maybe looking for the plugin "Printer settings":  https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/PrinterSettingsPlugin

CCS86, 2020-03-18:
I didn't realized it had been moved into a plugin

CCS86, 2020-03-18:
I installed that plugin and restarted Cura, but no new sidebar has appeared. 

[image]

DivingDuck, 2020-03-18:
Set Setting visibility to all. 

[image]

I use an additional plugin for all settings. Maybe its helpful for you too. It is called Sidebar Gui: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/SidebarGUIPlugin

CCS86, 2020-03-19:
I had misread the description of the plugin. I see that it added a new category to the settings sidebar.

Thanks guys!

closed 2020-03-19 by CCS86