Ultimaker/Cura #7297: Coasting: Add Flow Rate Coefficient

CCS86, 2020-03-18:
For bowden based printers, the experimental coasting feature is quite useful.

However, in its current implementation, it is hard to get a consistent result from it. A coasting setting that works beautiful at 45 mm/s print speed, will leave gaps at 25 mm/s.

I attribute this to the lack of a speed based coefficient. The phenomenon we are trying to control is extruder pressure, which is directly linked to volumetric flow rate, and that is linked to print head speed.

I think ideally there would be a coefficient in the calculation for volumetric flow rate, so that the coasting volume varied linearly with flow rate, so that it was 0 mm^2 at 0 mm^3/s. You could define that line using the [0, 0] point, and a user defined [x mm^3] at [y mm^3/s].

Ghostkeeper, 2020-03-19:
So what should the coasting volume become? Shall we say, the amount of material that would be extruded in one second?

A formula for that can be devised without adding any new setting: `=speed_wall_0 * layer_height * wall_line_width_0`. Of course the inner wall can be different from the outer wall, but it won't be very significant and you'd want to optimise the outer wall more than the inner wall. These formulas are not user-definable though, just by the profile authors. To do more we'd have to have a new setting that would change the unit (like Wall Thickness vs. Wall Line Count).

CCS86, 2020-03-20:
I like that idea. 

Can I experiment with creating a formula for that field? I know that I can't directly input it into the field. But can I export a profile of mine and edit the file, then import it?

fieldOfView, 2020-03-20:
You don't have to import and export. Instead, go to Help->Show configuration folder. Inside that folder, see the `user` folder, which contains .cfg.ini files for each printer you have configured. You can change setting values there, like so:

`wall_thickness = =speed_wall_0 * layer_height * wall_line_width_0`
(note the double `=`)

Before you edit that file though, close Cura. Once done editing, restart Cura.

CCS86, 2020-03-20:
I'm getting a red field after making that edit. Here is my definition:

`coasting_volume = = speed_wall_0 * layer_height * wall_line_width_0 * .04`

CCS86, 2020-03-21:
When I take out the '.04' term, the field is no longer red and has a value of 1.8, which would be correct. However, when I change wall line width, wall speed, or layer height; the value does not change.

fieldOfView, 2020-03-21:
Check the logs. I'm not sure, but I think it is not allowed to have a space after the second `=`.

Ghostkeeper, 2020-03-24:
> However, when I change wall line width, wall speed, or layer height; the value does not change.

This is due to a known bug. If a setting relation is not established in a definition file but in a profile (such as the `user` profile) then it won't automatically update. If the formula is written in a .def.json file it should update correctly. You wrote the formula correctly; it's on our side.