Ultimaker/Cura #7300: [4.5.0] Machine settings unreadable

njibhu, 2020-03-18:
**Application version**
4.4.0 and 4.5.0 (Did not test earlier)

**Platform**
Archlinux, both from community repo and AppImage.
KDE/Plasma 5.18.3 with default light theme, Xorg.

**Reproduction steps**
1. Open the machine settings

**Screenshot(s)**
https://imgur.com/a/ncdvTy7

**Actual results**
The settings previously set cannot be read, the text is offset up from the input field.

**Expected results**
Can read the previously entered settings.

**Log file**
```
2020-03-18 23:44:15,620 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/ComboBoxWithOptions.qml:74: TypeError: Cannot call method 'match' of undefined
2020-03-18 23:44:15,623 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/ComboBoxWithOptions.qml:74: TypeError: Cannot call method 'match' of undefined
2020-03-18 23:44:15,626 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/ComboBoxWithOptions.qml:74: TypeError: Cannot call method 'match' of undefined
2020-03-18 23:44:15,629 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/qml/UM/Dialog.qml:86:13: QML Loader: Binding loop detected for property "active"
2020-03-18 23:44:15,632 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:43:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,636 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:43:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,639 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:43:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,642 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/ComboBoxWithOptions.qml:39:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,645 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/SimpleCheckBox.qml:35:30: Unable to assign [undefined] to QString
2020-03-18 23:44:15,648 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/SimpleCheckBox.qml:35:30: Unable to assign [undefined] to QString
2020-03-18 23:44:15,650 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/SimpleCheckBox.qml:35:30: Unable to assign [undefined] to QString
2020-03-18 23:44:15,651 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/ComboBoxWithOptions.qml:39:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,653 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:43:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,655 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/PrintHeadMinMaxTextField.qml:56:16: Unable to assign [undefined] to QString
2020-03-18 23:44:15,657 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/PrintHeadMinMaxTextField.qml:38: SyntaxError: JSON.parse: Parse error
2020-03-18 23:44:15,658 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:43:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,660 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/PrintHeadMinMaxTextField.qml:56:16: Unable to assign [undefined] to QString
2020-03-18 23:44:15,662 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/PrintHeadMinMaxTextField.qml:38: SyntaxError: JSON.parse: Parse error
2020-03-18 23:44:15,663 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:43:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,665 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/PrintHeadMinMaxTextField.qml:56:16: Unable to assign [undefined] to QString
2020-03-18 23:44:15,667 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/PrintHeadMinMaxTextField.qml:38: SyntaxError: JSON.parse: Parse error
2020-03-18 23:44:15,668 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:43:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,670 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/PrintHeadMinMaxTextField.qml:56:16: Unable to assign [undefined] to QString
2020-03-18 23:44:15,672 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/PrintHeadMinMaxTextField.qml:38: SyntaxError: JSON.parse: Parse error
2020-03-18 23:44:15,674 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:43:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,675 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/ComboBoxWithOptions.qml:39:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,677 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/SimpleCheckBox.qml:35:30: Unable to assign [undefined] to QString
2020-03-18 23:44:15,679 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/GcodeTextArea.qml:27:30: Unable to assign [undefined] to QString
2020-03-18 23:44:15,680 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/GcodeTextArea.qml:27:30: Unable to assign [undefined] to QString
2020-03-18 23:44:15,694 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:43:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,696 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:43:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,697 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:43:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,699 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:43:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,701 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:43:34: Unable to assign [undefined] to QString
2020-03-18 23:44:15,703 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/GcodeTextArea.qml:27:30: Unable to assign [undefined] to QString
2020-03-18 23:44:15,705 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///tmp/.mount_UltimaQnOyFf/usr/bin/resources/qml/MachineSettings/GcodeTextArea.qml:27:30: Unable to assign [undefined] to QString
```

**Additional information**
Change cura theme or QT scaling variable doesn't solve the issue.

nallath, 2020-03-20:
I don't have the issue myself, so I'm not sure if my changes actually solve it.

smartavionics, 2020-03-20:
That change made no difference for me although what I see isn't as bad as the example above.

Mosher23, 2020-03-21:
same on Mac OS X Settings window is also not resisable
[image]

njibhu, 2020-03-22:
> I don't have the issue myself, so I'm not sure if my changes actually solve it.

Will try the patch this afternoon and will get back to you

> same on Mac OS X Settings window is also not resisable
> [image]

I actually can resize the window to make the G-Code input wider, I do not think that the bug is related. Though changing the window size for me doesn't change anything on my bug

njibhu, 2020-03-22:
@nallath tested by rebuilding an appImage, the problem is still here..

njibhu, 2020-03-23:
@nallath Found a fix and a way to reproduce it.
Using as a system font Noto Sans 10px, the bug is there. If i switch to Liberation Sans 11pt, it's gone. (with Liberation 10px, the input is still offset, but at least readable).
So it's definitely linked to font size

Ghostkeeper, 2020-03-24:
Not reproducing the problem with that on Ubuntu when running from source, but it is affected by it. This is my machine settings dialogue for Ender 3 using Noto Sans 10px:

[image]

And here is the dialogue using Liberation Sans 11pt:

[image]

But when you're using the AppImage, it should always use Noto Sans 10pt.

Pimentoso, 2020-03-27:
+1 for the problem on MacOS, please make this dialog resizable. Catalina 10.15.4, cura 4.5

nallath, 2020-03-30:
The dialog has been made resizable already.

Mosher23, 2020-03-30:
> The dialog has been made resizable already. 

But not on Mac OS or not in 4.5

Pimentoso, 2020-03-30:
> The dialog has been made resizable already.

Doesn't look like it. Also tried to resize it using a window manager (Rectangle) but it still doesn't change.

nallath, 2020-03-30:
It wasn't done in 4.5 if I remember correctly.

jlsjonas, 2020-03-31:
Can this be patched in? @nallath
Or at least a default height fix to make the start & end g-code windows readable on Mac?

Ghostkeeper, 2020-04-01:
The window is resizable in Cura 4.5 in Linux. If we made a change in 4.6 it must've been to fix something in MacOS only.

I also remember this coming back up recently, but I can't find the commits or any change that we made to fix it, and don't remember that we made any special handling for MacOS then.

jlsjonas, 2020-04-02:
@rburema might know more? @Ghostkeeper

Ghostkeeper, 2020-07-06:
This should be fixed in Cura 4.7, through this PR: https://github.com/Ultimaker/Cura/pull/7883

closed 2020-07-06 by Ghostkeeper