Ultimaker/Cura #7301: Cura 4.50 The overlap of lines.

GeserIO, 2020-03-19:
**Application version**
[4.50]
**Platform**
Windows10  x64

**Printer**
DeltaBot

**Reproduction steps**
1. Load the following model:
[Test.zip](https://github.com/Ultimaker/Cura/files/4352167/Test.zip)
2. Make sure that you have a 0.4 nozzle and that Fill Gaps Between Walls is set to everywhere.
**Screenshot(s)**
[image]
[image]

**Actual results**
When printing parts with a width of 2 mm, 2 outer walls, 4 layers of the bottom. The "Fill gaps between walls" and "bottom thickness lines" are overlaid. As a result, an increased pressure is created in the extruder and a huge drop of plastic is squeezed out when moving to the next layer.

**Expected results**
Cura should not be filling an already existing bottom lines. The width of these walls on the model is exactly 2 mm. This should be normally filled with 5 wall lines with no gaps needing to be filled.

**Project file**
[CURA test.zip](https://github.com/Ultimaker/Cura/files/4352171/CURA.test.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4352174/cura.log)

**Additional information**
If I enable "Extra skin wall count", three lines are overlaid.

rburema, 2020-03-20:
Could you save (not export) the project and attach it here? (Zip or rename to get github to accept it.)

Because if I follow the reproduction steps as-is, slicing on the DeltaBot doesn't result in a slice with the inner yellow walls for me.

GeserIO, 2020-03-22:
[CURA project.zip](https://github.com/Ultimaker/Cura/files/4365240/CURA.project.zip)

mahtDFR, 2020-03-23:
Thanks for the report, we have discussed and won't be devoting our time to this as it seems to occur for a specific third party printer.

Ghostkeeper, 2020-03-24:
Reduce the Skin Overlap to 0%. It's doing this because it creates 0.02mm overlap between the skin and the walls (for better strength). This makes your skin's area a bit bigger than what you are expecting.

closed 2020-03-24 by Ghostkeeper