Ultimaker/Cura #7302: Keyboard shortcuts do not function on OS/X for the popup dialog boxes

gitcnd, 2020-03-19:
**Application version**
v4.5.0
**Platform**
Mac OSX (with external magic keyboard 2.0 if that is relevant)

**Printer**
Anycubic Chiron

**Reproduction steps**
1. Open anything (irrelevant step)
2. Press Command-N to start a new project (alternatively - do anything that causes any popup - like saving a file that already exists etc).
3. The popup shows asking "Are you sure you want to start a new project" - with "No" or "Yes" options, with the N and the Y underlined to indicate they are available keyboard shortcuts.
4. Press Y  (alone, and/in in combination with command/alt/control/shift) - nothing happens.

**Screenshot(s)**
http://chrisdrake.com/shortcut_fail.png

**Actual results**
Nothing happens

**Expected results**
Should obey the "Y" or "N" keyboard keys when pressed

**Project file**
N/A

**Log file**
N/A

**Additional information**
I'm using an external keyboard, although this is probably irrelevant, and even if not, still needs fixing anyhow.

Vandrasc, 2020-03-19:
Hi @gitcnd , indeed I can confirm it doesn't work and never did actually, I will bring it up to our developers. Thank you!

ninovanhooff, 2020-03-23:
Relevant part of the documentation:

On certain widgets, using '&' in front of a character will automatically create a mnemonic (a shortcut) for that character, e.g. "E&xit" will create the shortcut Alt+X (use '&&' to display an actual ampersand). The widget might consume and perform an action on a given shortcut. On X11 (Linux) the ampersand will not be shown and the character will be underlined. On Windows, shortcuts are normally not displayed until the user presses the Alt key, but this is a setting the user can change. **On Mac, shortcuts are disabled by default. Call qt_set_sequence_auto_mnemonic() to enable them. However, because mnemonic shortcuts do not fit in with Aqua's guidelines, Qt will not show the shortcut character underlined.**

https://doc.qt.io/qt-5/qshortcut.html

This person using qt 5.11 does not see any underlinings:

https://forum.qt.io/topic/93463/qml-messagedialog-button-order-reversed-on-windows-linux-macos

@Vandrasc could you try to reproduce this on our experimental qt 5.14 build?

gitcnd, 2020-03-23:
Make the shortcuts work first.

Deal with missing accelerator key indicators later.

To remove all doubt - they do not work at all, no matter with ALT or any other modifier keys pressed.

closed 2024-12-07 by github-actions[bot]