Ultimaker/Cura #7303: Separeted x-y distance for supports from build plate and model

Liger0, 2020-03-19:
**Is your feature request related to a problem? Please describe.**
I want to have a big gap if the support starts from the build plate, but this way the distance is too much for the supports starting from the model, which fail.
If I use a little gap, the supports are hard to remove if they start from the build plate.

**Describe the solution you'd like**
Separate x-y distance from model for the supports that start from the build-plate and the ones starting from the model.

Ghostkeeper, 2020-03-23:
A couple of questions before we consider this:
* Why does it matter for the X/Y distance whether the support started from the build plate or rests on the model? It feels like you're just trying to filter out support for your particular print.
* What happens if a piece of support partially starts from the build plate and partially rests on the model?
* What happens when the adjusted X/Y distance when resting on the model is greater than when resting on the build plate, and this extra X/Y distance causes the support to no longer rest on the model but just on the build plate? This is a circular dependency.

Liger0, 2020-03-23:
The problem I was having is that both I have supports starting from the build plate all around the model hard to remove, and yet I have NEVER had supports starting from the model work, they always failed printing in the air.
I use even 3mm for x-y distance to make the supports far and 0.1mm as minimum x-y distance.
I was thinking that it would be possible to make the situation easier but I am a bit unsure.
The support, if starting from the model, should have the model itself support it better.

Ghostkeeper, 2020-03-24:
Reducing the Support Bottom Distance is not a solution?

Liger0, 2020-03-24:
I tried 0 layers but it still makes no difference.
I may try to reformulate the suggestion after some little thoughts.
It shouldn't be really dependant on the support starting from the model or the build plate, which may do things worse, because if a model starts from a flat part of the model, it gets near the model and is hard to remove even though it is not required.
A possibility would be to consider the first layer of the support and, if it's too little, consider the angle it produces with the eventual next layer support, IF it also starts from the model, this way we can know in advance if it will be at an overhang and has the possibility to fail, without making the supports starting from a flat spot too much near the model.
Another option is to consider how big is the base layer. If it's too little (value in the profile), it will have a 0 layer gap. If it's bigger than that value, it will have a layer (or more as per profile) of gap.
This would help even on flat spot supports that have to support dots, altough this doesn't help in the overhang case to me.

closed 2020-06-19 by no-response[bot]