Ultimaker/Cura #7304: [4.5] Comments removed from Start/End G-codes when opening from .3mf

Block137, 2020-03-19:
**Application version**
4.5

**Platform**
Win10

**Printer**
Custom

**Reproduction steps**
1. Put some G-code descriptions as new line comments in Start/End G-codes (; is the 1st character of a line)
2. Save a project as .3mf
3. Open .3mf file from step 2
4. Check Start/End G-codes again

**Actual results**
MOST comment only lines (lines starting with ";") were removed. Comments following actual G-codes (`G28 ;home` for example) remain unchanged.
Commented first line didn't get removed.

**Expected results**
Whole G-code scripts remain unchanged.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Additional information**
This is similar to #6689 but different condition.

Vandrasc, 2020-03-19:
Hi @Block137 I am not able to reproduce this, my comments at the beginning of start/end g-code starting with (;) are still visible after loading the project file. Did you use any other characters? Do you still have Marlin as g-code flavor?

rburema, 2020-03-19:
@Vandrasc  I think you may need to remove the printer in between the steps, I got it to reproduce if I did that (and had multiple ';' in each start/end gcode).

Block137, 2020-03-19:
@Vandrasc the very first line isn't affected, any other commented line will be removed. I'm using Marlin (volumetric)

Vandrasc, 2020-03-20:
Nice! Oke good thing it is reproducible, that makes it easier for us to fix. I will bring it up with our devs, thank you for reporting it!

fieldOfView, 2020-03-20:
@Vandrasc, there is now a PR to bring up too.

rburema, 2020-03-20:
PR was really small & made sense so I merged it as a 5 minute fix.

For internal QA: See jira ticket CURA-7315

PR #7311 (2020-03-20): Fix loading comments in gcode snippets from 3mf projects
This PR fixes loading of comments in start and end gcode snippets when loading in a project, by setting the appropriate option for the ConfigParser to not regard comments in the ini structure as comments but to read them as text instead.

This PR fixes #7304
commit 7f6a39d7 (2020-03-20): Fix loading comments in gcode snippets from 3mf projects

Fixes #7304
closed 2020-03-20 by rburema