Ultimaker/Cura #7307: Don't include scripts when installing v4.5

Alvgon, 2020-03-19:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

When v4.5 is installed, Cura includes postprocessing scripts that change extruder temp every 25 layers. 

Took me some time and half a kg of filament to understand what it was happenning. 

Please, please ... don't include active postprocessing scripts at installation time !!

fieldOfView, 2020-03-19:
Any postprocessing scripts are not added "out of the blue" upon installation, but they are copied over from a previous configuration, along with the printer configuration and other settings.

fieldOfView, 2020-03-19:
To mitigate your problem, Cura 4.6 will show more clearly the number of postprocessing scripts that are active, like this:
[image]

Alvgon, 2020-03-20:
> 
> 
> Any postprocessing scripts are not added "out of the blue" upon installation, but they are copied over from a previous configuration, along with the printer configuration and other settings.

Thanks for answering so quickly. If the scripts are not part of v4.5, then they were from v4.5 beta.
I upgraded from v4.0 that didn't have those scripts, to v4.5 beta.; and following day to v4.5 standard. Took me a while to understand what was happenning with the extruder temp.

If somebody else took the same version path, probably will have those (asumed for testing purposes) scripts active.

fieldOfView, 2020-03-20:
> If somebody else took the same version path, probably will have those (asumed for testing purposes) scripts active.

Not unless they manually activated those scripts, or imported a project with those scripts activated. Cura does not (and has not, in any version) activate postprocessing scripts on its own.

Ghostkeeper, 2020-03-24:
I took the following reproduction steps:
* Cleaned any previous Cura configuration.
* Start Cura 4.0.0.
* Add a Creality Ender 3 printer.
* Close Cura 4.0.0.
* Start Cura 4.5.0.
* Load a model (otherwise the post-processing scripts button is not visible).

No post-processing scripts have been loaded then, as can be seen by there not being this wrench and hammer icon near the slicing panel: 
[image]

There are also no post-processing scripts active when opening the actual dialogue in Extensions.

Maybe you loaded a project file that had the post-processing scripts in them?

Alvgon, 2020-03-24:
Hace hoy tries v4.5 beta?

Ghostkeeper, 2020-03-26:
I did now, though supporting that beta is not a concern of course. But I get the same result:

[image]

The code for the version upgrade didn't change during the beta.

Alvgon, 2020-03-26:
Ok then, lets close this. Many thanks for your attention to this. BRs.

closed 2020-03-26 by Alvgon