Ultimaker/Cura #7308: Active/inactive flag every single post processing script

Alvgon, 2020-03-19:
**Is your feature request related to a problem? Please describe.**
When v4.5 is installed, a series of post processing scripts is active, decreasing 10º the extruder temp, every 25 layers (it seems a v4.5 beta residue)

**Describe the solution you'd like**
I'd like to have a way of, keeping the post processing scripts defined, activate them or not with a flag in each one.

**Describe alternatives you've considered**
An alternative could be an activation flag affecting all post processing scripts as a block.

OR additionally have a way of nesting scripts in processing blocks.That would be even nicer.

**Affected users and/or printers**
Benefits are clear, may be advanced users will make a better use of it.

**Additional context**
(Add any other context or screenshots about the feature request here.)

Ghostkeeper, 2020-03-24:
You can activate and deactivate them in this window, right?
[image]
Activate them with "Add a script", deactivate them with the X button. They always stay defined.

Is there something I'm missing? I don't think I understand your feature request.

fieldOfView, 2020-03-24:
By removing a script from the list of added scripts, you also remove its configuration. I think the request is for an additional layer of activation/deactivation so you can temporarily disable a script. Personally I think that will only make things more confusing.

Alvgon, 2020-03-24:
I coldn't have any it better!

Ghostkeeper, 2020-03-25:
So to implement this we'd need to have sets of post-processing script configurations (profiles, if you will) that you can save and re-load.

I'm agreeing with fieldOfView's opinion here. I think this would be more confusing than useful to most people.

As a workaround, you can save and load post-processing script configurations through the project files (FIle -> Save...).

closed 2020-03-25 by Ghostkeeper

pto199, 2022-03-23:
I'm new to Cura and just getting around and find it crazy that this wasnt implemented.  Its such an easy and needed addon and I find it confusing that there is no option to enable/disabled post processing scripts.

I have a script "search and replace" which I use for some prints and not for others.  Everytime I have to remove it and then when I need to use it again on a later print, I need to figure out what text it is I need to re-enter.  What a hassle.  Why cant there just be checkboxes next to each script and if theres a check, its enabled.  No check, its disabled.  Simple and non confusing.

shibedroid, 2024-07-01:
I agree, I am using an old FlashForge Creator Pro 2 and boy it's a doozy. Since there is no native support for this printer, I have to postprocess every print I slice, substituting some code to make the x3g work. Currently I have to substitute 6 fragments of code but to add insult to injury, the code substitution is different if the print has supports or brims so it's a real hassle having to delete each code substitution and then adding them again when they could easily just be inactivated and reactivated when needed.

WhyIcanBuyMe20atAnyMarket, 2024-08-01:
If you use the same script settings for every time that you use that script, you could edit the script's default values:
[image]
The scripts are found (on windows; I don't know about any other OS) in `C:\Program Files\UltiMaker Cura 5.7.2\share\cura\plugins\PostProcessingPlugin\scripts` or somewhere similar for other versions.

Unfortunately a limitation of this is that I can't find a way to have line feeds (`\n`) as a default value, having tried both that and `\\n`.
You could instead customise the script such that (for instance for 'search and replace' as pto199 suggested) the setting value is removed and coded into the script, like so:
[image]
Multiple scripts, or multiple parts of a single script which are enabled/disabled by a setting, can then be used to choose between a predefined set of values. The example below shows a possible modification of 'search and replace' with a list of three pre-defined search strings, based on whether a new parameter says 'support', 'brim', or 'infill'.
`# Copyright (c) 2017 Ghostkeeper
# The PostProcessingPlugin is released under the terms of the LGPLv3 or higher.

import re # To perform the search and replace.

from ..Script import Script

class SearchAndReplaceV11(Script):
    """Performs a search-and-replace on all g-code.

    Due to technical limitations, the search can't cross the border between
    layers.
    """

    def getSettingDataString(self):
        return """{
            "name": "Search and Replace v11",
            "key": "SearchAndReplaceV11",
            "metadata": {},
            "version": 2,
            "settings":
            {
                "mode":
                {
                    "label": "Operation mode",
                    "description": "Options are 'none' (optional), 'all', 'support', 'brim', or 'infill'. Seperate with commas and optional spaces.",
                    "type": "str",
                    "default_value": "none"
                },
                "replace":
                {
                    "label": "Replace",
                    "description": "The search text will get replaced by this text.",
                    "type": "str",
                    "default_value": ""
                },
                "is_regex":
                {
                    "label": "Use Regular Expressions",
                    "description": "When enabled, the search text will be interpreted as a regular expression.",
                    "type": "bool",
                    "default_value": false
                }
            }
        }"""

    def _replaceAll(self, local_data, search_string):#moved to a new function so that we can call it multiple times using different variables
        #call using self._replaceAll(local_data, search_string)
        
        if not self.getSettingValueByKey("is_regex"):
            search_string = re.escape(search_string) #Need to search for the actual string, not as a regex.
        search_regex = re.compile(search_string)

        replace_string = self.getSettingValueByKey("replace")

        new_data = []
        for layer in local_data:
            new_data.append(re.sub(search_regex, replace_string, layer)) #Replace all.
            
        return new_data

    def execute(self, data):
        mode_string = self.getSettingValueByKey("mode")
        mode_list = mode_string.split(',')#seperate the diferent mode values
        setting_index = -1
        for setting in mode_list:
            setting_index += 1
            mode_list[setting_index] = mode_list[setting_index].strip()#remove whitespace around the mode input
        if 'All' in mode_list:
            mode_list = ['support','brim','infill']
        if 'support' in mode_list:
            data = self._replaceAll(data, "STRING TO SEARCH FOR WITH SUPPORT")#EDIT THIS!!!
        if 'brim' in mode_list:
            data = self._replaceAll(data, "STRING TO SEARCH FOR WITH BRIM")#EDIT THIS!!!
        if 'infill' in mode_list:
            data = self._replaceAll(data, "STRING TO SEARCH FOR WITH INFILL")#EDIT THIS!!!
        return data
`

PLEASE NOTE that if you modify any post-processing script, or create a new one, you should also move it into 'scripts' sub-folder of the configuration folder. The configuration folder is found at `Help→Show configuration folder` (drop-down menu in cura). This move is such that the updated script is carried over if you update cura.