Ultimaker/Cura #7313: Using gradual infill and object for parameters overlaps inside creates more infill under bottom object (set to 0)

texscalper, 2020-03-20:
**Cura 4.5**

**Platform**
Windows 10

**Printer**
Sunlu S8 (CR10S), Kossel LP (Delta)

**Reproduction steps**
1. gradual infill 9%, steps 3, mm step 3
2. add object modify settings for overlap inside mesh
3. set parameters overlap : top 0 layers, bottom 0 layers, wall number 0, infill density 0%

**Actual results**
Sliced mesh shows gradual infill under bottom of inner void object (cube)

**Expected results**
Standard infill should be created under a void object with bottom set to 0

Ghostkeeper, 2020-03-23:
This is a consequence of how the infill meshes work right now. They subdivide the infill into different parts. The gradual infill algorithm looks for where the infill gets close to the top side of its infill volume, which is indeed interrupted by the infill mesh. Not nice for trying to get full control over the infill.

We've discussed this but we've had to defer this issue. The number of users that use this combination is not big enough to warrant spending time on this now. We may come back to this issue later.

texscalper, 2020-04-04:
ok, I think in coding view, it's just a matter of reading if bottom layers of the other object is 0 or not..