Ultimaker/Cura #7314: skr e3 mini, dont print via USB cable

jpokorny12, 2020-03-20:
Hello
You cannot print from a Cura 4.5 over a usb cable. The board in the printer is e3 mini.
Thank you

Ghostkeeper, 2020-03-24:
Yes, you can. I've tried it. My board is an UltiMainBoard designed for the Ultimaker Original. Are you sure it's not the board or the cable that's at fault?

Ghostkeeper, 2020-03-24:
What are your reproduction steps and what are you seeing? Is it not recognising the printer via the cable? Is it refusing to send g-code when you press "print via USB"? Is it accepting the g-code but printing dildos instead of your actual print? Could you fill out the [bug template](https://github.com/Ultimaker/Cura/blob/master/.github/ISSUE_TEMPLATE/bug-report.md) that you erased when you reported this bug?

jpokorny12, 2020-03-24:
Cable and everything is well set up. The printer can be operated via Repetier. Cura will not show the ability to print via USB. Only when the printer is connected and the zur is turned on, a whistle will sound like an alarm.
I found a post where someone else solves it. In the second post is a link to the video.
https://community.ultimaker.com/topic/30043-does-not-connect-with-ender3-and-skr-mini-e3-board/

Ghostkeeper, 2020-03-25:
Eek, static electronics should not make that sound!

We can't remove those lines from Cura though. They are necessary to be able to program the firmware from Cura, i.e. its firmware update functionality.

Did you opt to install the Arduino drivers when you installed Cura?

Why would connecting a programmer cause the PCB to start whistling?

Ghostkeeper, 2020-03-25:
> Why would connecting a programmer cause the PCB to start whistling?

#BrandNewSentence

jpokorny12, 2020-03-25:
When I coment this line, and when I turn on the Cura the board doesn't whistle.

Yes i install arduino drivers

c : \ Program Files \ Ultimaker Cura 4.3 \ plugins \ USBPrinting \

Solution 1:

Line 33 # try:
Line 34 # programmer.connect (self._serial_port)
Line 35 # serial = programmer.leaveISP ()
Line 36 # except ispBase.IspError:
Line 37 # program.close ()

jpokorny12, 2020-03-27:
I bring good news. I completely reinstalled the Cura. The whistle did not stop. I opened the python again and commented on the lines already mentioned. Cura recognized the printer. but I haven't tried printing yet. The printer is now printing long prints. Then I'll try

closed 2020-06-19 by no-response[bot]