Ultimaker/Cura #7317: [4.5.0] Spiralize Outer Contour displays wrong Wall Thickness

mathiasvr, 2020-03-20:
**Application version**
4.5.0

**Platform**
macOS Mojave

**Printer**
Ender 3

**Reproduction steps**
1. Use any default profile and enable *Spiralize Outer Contour*
2. Wall Line Count is changed from 3 to 1, but Wall Thickness remains 1.2mm

**Expected results**
Since Wall Line Count of 1 is correct, the Wall Thickness should be 0.4mm according to the definitions.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Additional information**
The model prints fine with a Wall Thickness of 0.4mm, but the setting of 1.2mm seems confusing.

Ghostkeeper, 2020-03-24:
The Wall Line Count overrides Wall Thickness, so the single wall is what is actually used. This is a natural problem with having multiple settings that essentially control the same thing. Myself I'd go for just having a single setting and just being able to change the units. But it's a bit late to change that now; I don't see that happening any more. There are thousands of profile in Cura that depend on the current behaviour and many more outside of Cura.

I can make it also update the parent setting. Maybe that's best.

closed 2020-03-24 by Ghostkeeper

mathiasvr, 2020-03-24:
Thanks!
In case of multiple settings it could be nice if the ones overridden where “grayed out” to show that they are not in effect.

Ghostkeeper, 2020-03-26:
They get greyed out automatically if all of their subsettings are overridden by the user, but in this case the subsetting (wall line count) is not overridden by the user. It's still on its default, but the default is a formula that adjusts itself based on whether Spiralize is enabled or not. It's a bit complex to make it grey itself out automatically in this case.

commit 8e324c59 (2020-03-24): Adjust wall thickness to 1 line if using spiralise

Except for printers that explicitly override it.

Fixes #7317.