Ultimaker/Cura #7319: Straight infill of full layer

Rendojack, 2020-03-21:
Need ability to prevent Cura slicer from leaving certain object parts for later infill. Let it go straight lines for full layer. Prusa slicer has this feature by default.

Illutration of issue:
https://imgur.com/a/fFaGwaB

smartavionics, 2020-03-21:
So how does Prusa slicer do it? Please post some more images that illustrate that.

Rendojack, 2020-03-21:
Prusa does a nice and smooth surface, then builds those towers on top of that.
Cura is starting to build towers inside of top layer, then leaves infill gaps to finish later. 

Images: https://imgur.com/a/XAppoSE
Stl for slicing: https://easyupload.io/flm2yn

smartavionics, 2020-03-21:
Increase the skin expand distance...

[image]

Rendojack, 2020-03-21:
Thanks. That is the answer. Didn't know this feature existed, even cura community on facebook weren't able to help with this.

closed 2020-03-21 by Rendojack