Ultimaker/Cura #7320: [4.5.0] Layer height Z-move change applied after "Z-Hop When Retracted" move

jcwebber93, 2020-03-21:
**Application version**
Cura 4.5.0

**Platform**
Windows 10

**Printer**
Custom FFF Profile (Can also use Ultimaker S5 profile for testing purposes)

**Reproduction steps**
1. Turn "Combing Mode" "OFF"
2. Enable "Z Hop When Retracted"

[image]

**Screenshot(s)**
[image]
[image]
[image]
[image]

**Actual results**
Layer "A"
1. Final print move of layer "A" completes
2. Filament retract prior to non-print move
3. Z-hop of 1.5mm because of selected "Z Hop When Retracted" and previous retract
4. Move to X and Y start of next layer "B"

Layer "B"
5. Layer count present as a comment
6. Another call to move to the X and Y start of layer "B" (identical values as #4) WITH an increase in Z based on layer print height
7. Z-hop of 1.5mm back to begin a print move

[image]

**Expected results**
Layer "A"
1. Final print move of layer "A" completes
2. Filament retract prior to non-print move
3. Z-hop of 1.5mm because of selected "Z Hop When Retracted" and previous retract
4. Move to X and Y start of next layer "B"

Layer "B"
5. Layer count present as a comment
6. Z-hop of 1.3mm back to begin a print move of layer "B"

[image]

**Project file**
[CFFFP_stra tall testPart1.zip](https://github.com/Ultimaker/Cura/files/4362907/CFFFP_stra.tall.testPart1.zip)

Attached as an actual ZIP, unzip to load into Cura (if desired)

**Additional information**
Around a layer change, the z-hop on retraction occurs, and then a secondary hop based on layer thickness occurs directly before a hop back to the new printing point, resulting in two very quick Z moves. I would expect this to be one move, not two. Z-hop -> move to new X+Y start ->Z-hop minus layer thickness change-> new layer, not the current system.

Ghostkeeper, 2020-03-24:
Indeed the currently expected behaviour for when a layer change happens during a Z hop is as follows:
[image]

This is to keep the logic a bit simpler. It's indeed an unnecessary move. It normally ends up with the layer change on the far side of the layer change, making it go back and forth.

I don't know if we really want to fix this considering the small size of the move. With a bit of a Z jerk of 1mm/s it should take less than 0.2s per layer. I'll put it up for discussion with the other programmers.

jcwebber93, 2020-03-25:
Ghostkeeper, many thanks for taking a look at this. I do agree that the time taken for this move is quite small (~2mins for essentially a four hour print for example). One of the issues I can see coming out of this - and it is coupled to properly tuning and selecting machine acceleration and jerk values, so not really an issue directly caused by Cura - could be skipping steps on the Z-axis when performing these moves with extreme jerk and acceleration values. I had once selected **outrageous** acceleration and jerk values and saw an ever-increasing air gap after layer changes, but at the time was unsure why it only occurred on layer changes. 

Your diagram does a good job of explaining the post, I'll keep a diagram in mind if I ever submit another issue. Thanks!

Ghostkeeper, 2020-03-25:
We've discussed this among the colleagues and decided that this is a valid problem, but the negative effect is so small that we won't give it any priority. So we'll defer this until we have more reason to prioritise it over other issues.

If anyone is looking into this, the solution is probably somewhere in CuraEngine where it determines the height to move to for the layer change. Take care not to make the code more complex there though.

closed 2023-06-24 by github-actions[bot]