Ultimaker/Cura #7321: Parameter error

gailulun, 2020-03-21:
cura4.x
windows10
When setting the parameters, this kind of error will always occur somehow after changing the "shell" related parameters.
[image]

gailulun, 2020-03-21:
Seems like there are no similar bug reports

smartavionics, 2020-03-21:
That's not an error, it's a warning and it won't stop you slicing. If the value is in error, it will be red rather than orange.

gailulun, 2020-03-21:
That weird parameter (9999 ....) appears automatically

gailulun, 2020-03-21:
And the top layer is 0.
I didn't set it on purpose

gailulun, 2020-03-21:
Sorry i didn't make it clear just now

smartavionics, 2020-03-21:
Could you please save the project file (File -> Save), zip up the .3mf file and attach to this issue, thanks.

smartavionics, 2020-03-21:
I know what is causing this, you have set the infill density to 100%. When you do that, the infill is replaced with bottom layers.

gailulun, 2020-03-21:
I think you are right, I just tried it. But this parameter is really weird, and I change it manually every time.

gailulun, 2020-03-21:
And this seems a bit unreasonable, cura will alarm the parameters set by herself

gailulun, 2020-03-21:
and there's another question,when printing multiple identical models at one time (similar to an array), cura has a tolerable threshold. When the number increases to a certain value, you can still slice at this time, and you cannot slice even if you increase one. This seems to have nothing to do with the performance of the computer.

Ghostkeeper, 2020-03-24:
For your second question, please make a separate issue and fill in the bug template.

Ghostkeeper, 2020-03-24:
The parameter is indeed changed on purpose according to the defaults. When you change the infill ratio to 100% we find that it better approaches 100% infill density by filling the entire model with skin rather than actually using infill. So we give it 999999 skin layers.

You can ignore the top layer warning unless your model is more than a million layers high. Normally having no top skin is problematic though so that's why it's giving a warning.

closed 2020-03-24 by Ghostkeeper